gepd/Deviot #4: Not detecting platformio

sticilface, 2016-01-03:
doesn't seem to figure it out.  

using mac, with platformio installed.. 

any help appreciated?

gepd, 2016-01-03:
When you run `platformio --version` in the terminal what is your output?

sticilface, 2016-01-03:
PlatformIO, version 2.7.0

gepd, 2016-01-05:
@sticilface I've pushed some changes in the code, I have not tested in the UNIX environment, but it should work. Let me know if that solve your issue.

sticilface, 2016-01-07:
Sorry that took me a while to get back to you.  
It works.  Good work.  

Next question.  OTA?  

the arudino environment does it via mDNS, but platformio uses specified IP addresses.  Are either of these options available?

ivankravets, 2016-01-07:
You can use mDNS with @PlatformIO too. See http://docs.platformio.org/en/latest/platforms/espressif.html#over-the-air-ota-update

gepd, 2016-01-07:
@sticilface It will have for sure! As I'm not have any OTA device in this moment I haven't had not the opportunity to implement this feature. This plugin has a little difference detecting the ports, but I'll try to add it ASAP, I'll let you know when this feature is available.

sticilface, 2016-01-07:
That is not quite what i meant.  The arduinoIDE has autodiscovery using mDNS, so the IDE can pick up the ESPs on the network, and they appear in the boards/ports list.  

@gepd cheers :)

closed 2016-03-14 by gepd

gepd, 2016-08-18:
@sticilface The auto discover feature is available now in version `1.2.0`