gepd/Deviot #6: Installation failing somehow

etaba, 2016-03-20:
Followed the instructions as far as i know:
cloned Deviot into the sublime packages directory
restarted sublime and it attempted installing platform io and deviot on its own
Claims the installation was successful... but no Deviot menu option, and this is what i see in the console:

startup, version: 2221 osx x64 channel: stable
executable: /Applications/Sublime Text 2.app/Contents/MacOS/Sublime Text 2
working dir: /
packages path: /Users/Taba/Library/Application Support/Sublime Text 2/Packages
settings path: /Users/Taba/Library/Application Support/Sublime Text 2/Settings
PackageSetup not required
catalogue loaded
found 5 files for base name Default.sublime-keymap
found 1 files for base name Default.sublime-mousemap
found 3 files for base name Main.sublime-menu
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/0_package_control_loader/00-package_control.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/0_package_control_loader/02-bz2.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/CSS/css_completions.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/comment.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/copy_path.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/delete_word.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/detect_indentation.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/duplicate_line.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/echo.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/exec.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/fold.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/font.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/goto_line.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/indentation.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/kill_ring.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/mark.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/new_templates.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/open_file_settings.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/open_in_browser.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/paragraph.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/save_on_focus_lost.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/scroll.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/set_unsaved_view_name.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/side_bar.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/sort.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/swap_line.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/switch_file.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/transform.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/transpose.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Default/trim_trailing_white_space.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Deviot/DeviotStarter.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Diff/diff.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/HTML/encode_html_entities.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/HTML/html_completions.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Package Control/1_reloader.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Package Control/2_bootstrap.py
Reloading plugin /Users/Taba/Library/Application Support/Sublime Text 2/Packages/Package Control/Package Control.py
plugin init time: 0.124853
loading bindings
loading pointer bindings
found 1 files for base name Default.sublime-theme
theme loaded
app ready
pre session restore time: 0.255007
using gpu buffer for window
using gamma: 2 (err: 6.85565)
Traceback (most recent call last):
  File "./sublime_plugin.py", line 205, in on_activated
  File "./sublime_plugin.py", line 154, in run_timed_function
  File "./sublime_plugin.py", line 204, in <lambda>
  File "./DeviotStarter.py", line 96, in on_activated
  File "./libs/PlatformioCLI.py", line 111, in __init__
  File "./libs/Paths.py", line 241, in getCurrentFilePath
AttributeError: 'NoneType' object has no attribute 'views'
wrote startup cache, added files: 1 orphaned files: 0 total files: 98 cache hits: 97
startup time: 0.360317 (package setup was not run)
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.6/lib/python2.6/threading.py", line 532, in __bootstrap_inner
    self.run()
  File "/System/Library/Frameworks/Python.framework/Versions/2.6/lib/python2.6/threading.py", line 484, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./libs/Install.py", line 97, in checkPio
  File "./libs/Install.py", line 256, in endSetup
  File "./libs/PlatformioCLI.py", line 544, in generateFiles
  File "./libs/PlatformioCLI.py", line 492, in saveAPIBoards
  File "./libs/PlatformioCLI.py", line 473, in getAPIBoards
RuntimeError: Must call on main thread, consider using sublime.set_timeout(function, timeout)

loaded 837 snippets
found 3 files for base name Main.sublime-menu
Package Control: Skipping automatic upgrade, last run at 2016-03-20 13:57:54, next run at 2016-03-20 14:57:54 or after

ivankravets, 2016-03-20:
@gepd it looks strange for me. Take a look at @etaba [comment]( https://www.reddit.com/r/arduino/comments/4b7ejs/new_sublime_text_plugin_for_arduino_development/d176hfw). It means that Python 2.7 should be called via `python`. I don't know what here is doing Python 2.6.

gepd, 2016-03-20:
Maybe it's because he is using Sublime Text 2, can you confirm this @etaba?

etaba, 2016-03-21:
Can confirm, using ST 2

gepd, 2016-03-21:
I've fixed this problem, try installing the plugin again (package control may take a while to update the plugin list, but not more than 2 hours from now)

etaba, 2016-03-22:
Thanks it works! Looking forward to using Deviot.

Out of curiosity, could you briefly describe what was the problem?

Thanks,
Eric

gepd, 2016-03-22:
Sure, ST2 has a different internal structure (compared with ST3) from a specific version ST2, plugins are only allowed to call API functions from the main thread, when you don't do it you have to fix it like the error said, using `sublime.set_timeout(function, timeout)`

closed 2016-03-22 by gepd