gepd/Deviot #7: Shortcut Symbols not showing correctly

AKBON, 2016-03-21:
Hi,

on a Windows 7 and XP system only rectangles are showing up instead of the correct symbols. Is it necessary to install any symbol library?

gepd, 2016-03-21:
I'll check this in a virtual machine, the symbols are the emojis from the system, it may have a solution in Win 7 but I'm not sure in XP, I'll back here when I finished the tests

gepd, 2016-03-27:
I've resolved this issue for Windows 7 in the last update (1.1.5) re-installing the plugin should change the icons, it won't be the same icons as the available in Windows 10 because W7 only have a few. 

Nothing to do in XP there is no icons available. I forgot to disable the bar, in the next release I'll do.

closed 2016-03-27 by gepd