gepd/Deviot #8: No new line in console output

AKBON, 2016-03-21:
Hi,

I have the problem, that in the console output everything goes into an endless string instead of beginning a new line for every statement. In your animated gifs everything looks fine. Is that a matter of preferences or what could be the reason?

Thx!

gepd, 2016-03-21:
what development board are you using?

AKBON, 2016-03-22:
Hi, it's not code depended. I mean the outputs of deviot in the console, while compiling or setting up something. That all goes into a long string with no line break in the sublime window. Console was the wrong word, I am not sure how it's called correctly^^

gepd, 2016-03-22:
You can said console, I'll understand,
To know where is the problem, I need to replicate the conditions of your project, without that information there is not much I can help. As you noted in the gift, with my setup all works fine.

AKBON, 2016-03-23:
It already hapens directly after installation of the plugin, so not much configuration has been done. My first try was uploading an example sketch to a NodeMCU ESP 8266. The problem is reproducable on other windows machines.

gepd, 2016-03-23:
can you provide me the link to the example sketch?

AKBON, 2016-03-23:
It occurs everytime, no matter what code is used and I'm not sure how the code could influence the console output. Consider the standard Arduino blink example, than the console output after building looks like this:

[ Deviot 1.1.3 ] Blink.ino
init_project_11:21:02success_took_5.96built_project_11:21:08

Another Issue: The names of the menu items are also not correct, they are like "menu_new_sketch" and so on.

Interesting: Problem seems only to exist on Windows, on a MAC the output and the menu items are both fine.

Could it be a problem with a parallel installation of Stino?

gepd, 2016-03-23:
Now I understand the issue, always adds the most information possible to understand the problem.

I've deviot installed in multiple O.S and I haven't that error, try disabling stino from the package control options and restarting ST.

you said It happens to you in all versions of windows or you tried a specific one?

AKBON, 2016-04-02:
Hi, it happens in XP an Win 7, that's where I could test it. I just made a fresh installation on my private laptop (Win 7) which never had contact to Stino and the problems are the same. Incorrect menu item names and no line break. Switching the language did not help.

gepd, 2016-04-02:
What is the default language in your machines? I'm thinking it can be because you're using a different language. (I'm just guessing)

Most problems with icons is because ST has problem with different OS, and don't support graphic buttons, I'm looking for a better solution.

AKBON, 2016-04-02:
System default language is German.

gepd, 2016-04-02:
Thanks, I know what can be the problem, it will be solved in the next release.

closed 2016-04-02 by gepd