gepd/Deviot #9: Error Creating the virtualenv

BadgerAAV, 2016-03-25:
During installation, an error occurs: "Error Creating the virtualenv"
[image]
 python "python-2.7.11.amd64" + add python.exe to path option

gepd, 2016-03-25:
Is that windows 10?

Please go to view > Show console 
and paste all the content of the console here.

BadgerAAV, 2016-03-25:
that windows 8.1 x64
DPI scale: 1
startup, version: 3103 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/Artem/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Artem/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Artem/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.197649
startup time: 0.244649
first paint time: 0.247649
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.profile_syntax_definition
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Python.syntax_test_python
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.02-bz2
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
plugins loaded
Package Control: Skipping automatic upgrade, last run at 2016-03-25 20:00:18, next run at 2016-03-25 21:00:18 or after
ignored packages updated to: ["Deviot", "Vintage"]
reloading Packages/User/Package Control.sublime-settings
reloading Packages/User/Preferences.sublime-settings
reloading Packages/User/Package Control.sublime-settings
ignored packages updated to: ["Vintage"]
reloading plugin Deviot.DeviotStarter
reloading Packages/User/Package Control.sublime-settings
reloading Packages/User/Preferences.sublime-settings
reloading Packages/User/Preferences.sublime-settings

gepd, 2016-03-25:
Please uninstall and install again the package and copy the console output only after the `Error creating the virtualenv` appears

BadgerAAV, 2016-03-25:
console output is the same as the last
(I deleted deviot package and installed again and then checked the console)

gepd, 2016-03-25:
Thanks I'll check here to see what is the problem, I'll back in few hours

BadgerAAV, 2016-03-25:
same error while try under sandboxie (try for clear install)

gepd, 2016-03-25:
Something weird is happening, I've create a virtual machine with win 8.1 64b and it installed Deviot correctly.

Please follow this steps:

To open the package folder go to `Preferences > Browse Packages`

* 1 - Open the file `Install.py` inside folder `Packages > Deviot > libs` 
* 2 - Go to the end and paste `print(stdout)` before `return (return_code, stdout)` and restart ST, 

* 3 - It should start to re-install the plugin if doesn't, please delete the folder `deviot` in `Packages > User` 

* 4 - After the installation fails, copy the output of the console here, it should have extra information available now.

BadgerAAV, 2016-03-25:
I add `print(stdout)`
[image]

 but I think I did it wrong, because the console error appears when ST start

Traceback (most recent call last):
  File "C:\Users\Artem\AppData\Roaming\Sublime Text 3\Packages\Deviot\DeviotStarter.py", line 29, in <module>
    from .libs.Install import PioInstall
  File "C:\Users\Artem\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs\Install.py", line 273
    print (stdout)
                 ^
TabError: inconsistent use of tabs and spaces in indentation

Please describe more how do I change a file.

gepd, 2016-03-25:
Python is very strict with the indentation, I'll let you the file edited [here](https://www.dropbox.com/s/00d546hf2wffc1h/Install.zip?dl=0)
extract it and replace in the `libs` folder

BadgerAAV, 2016-03-25:
now new errors on startup ST:

reloading plugin Deviot.DeviotStarter
Traceback (most recent call last):
  File "C:\Users\Artem\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs\Install.py", line 28, in <module>
    from .libs.I18n import I18n
ImportError: No module named 'Deviot.libs.libs'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Artem\AppData\Roaming\Sublime Text 3\Packages\Deviot\DeviotStarter.py", line 29, in <module>
    from .libs.Install import PioInstall
  File "C:\Users\Artem\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs\Install.py", line 30, in <module>
    from urllib2 import Request
ImportError: No module named 'urllib2'

gepd, 2016-03-25:
Sorry about that, please download the file again I've edited it.

BadgerAAV, 2016-03-25:
[log.txt](https://github.com/gepd/Deviot/files/190246/log.txt)

new log with strange error

gepd, 2016-03-26:
Reading about the Error 5 in python, It seems to be related to permissions

Are you in a account with administrator privileges?
If you have an antivirus please disable it and try to installing the plugin again

closed 2016-04-12 by gepd

aniston, 2016-06-13:
@BadgerAAV if you are on windows i guess you are missing 'virtualenv' , if so open a command prompt and use pip to install it with 
`pip install virtualenv`

stanleyseow, 2017-02-03:
Hi, I hv this same error again ..

20:17:45 Downloading necessary files...
20:17:46 Extracting files...
20:17:46 Installing PlatformIO...
20:17:46 Error Creating the virtualenv

I've 2 computers, the one with Win7 32bit install OK, but the one with Win7 64bit have the above problem..
SublimeText is both on 3162
I tried with admin and disable anti-virus, same results...

How do I produce the installation logs ?