gepd/Deviot #10: Multiple .pde files

thomsna, 2016-03-27:
I was previously using Stino, but want to give Deviot a try.

I have a project folder with multiple .pde files that are alphabetically sorted, that compiled fine on Stino.

Now, I have out of scope errors. How are multiple files handled in Deviot?

gepd, 2016-03-27:
It should work without modification, do you have any place where I can download the files and test it?

thomsna, 2016-03-27:
Here is a sample project: https://dl.dropboxusercontent.com/u/7038123/pde-test.zip

It's using this library: https://github.com/gioblu/PJON

gepd, 2016-03-31:
Sorry for the delay @thomsna 

I'll copy the answer from the [PlatformIO](https://github.com/platformio/platformio) team

>> I think that approach with alphabetically sorted .pde files is quite limited, but you always can include one .pde file into other via #include "file.pde" with "include guard"

a-test.pde:

```
#ifndef _A_TEST_PDE_
#define _A_TEST_PDE_
#include <PJON.h>
#endif  //_A_TEST_PDE_
```

z-main.pde:
```
#include "a-test.pde"

PJON network(12); // Bus connection to pin 12

void setup() {

  PJON network(12, 45); // Bus connection to pin 12

  network.send(44, "B", 1, 1000000);
  // Send to device 44, "B" content of 1 byte length every 1000000 microseconds (1 second)
}

void loop() {
  network.update();
}
```

If you try that example it's working now.

Additionally, the PlatformIO team has included the [PJON](http://platformio.org/lib/show/245/PJON) library in his library manager. So now you can go to `Deviot > Search Library` search `PJON` and install the library with a few clicks.

thomsna, 2016-03-31:
Your solution works, thanks.
You're right, it's much more flexible that way.

gepd, 2016-03-31:
I'm happy to help you

closed 2016-03-31 by gepd