gepd/Deviot #12: Can't select serial port

smartsystemdesign, 2016-03-30:
Deviot was working fine on my Windows 10 64bit machine and for some reason, I removed the preferences file.  After that, I have not been able to select a serial port.  The serial ports menu item is empty, even though I see COM3 in Device Manager.  I can still select and upload programs through the Arduino IDE.
[image]

gepd, 2016-03-31:
can you please go to `View -> Show Console` and paste here the output of your console?

smartsystemdesign, 2016-03-31:
The last line below (Unable to parse command: ), was added to the console when I clicked on Serial Port(s) in the Deviot menu.

Console output:

DPI scale: 1
startup, version: 3103 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/NHopkins/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/NHopkins/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/NHopkins/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.116366
startup time: 0.146366
first paint time: 0.146366
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.profile_syntax_definition
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Python.syntax_test_python
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.02-bz2
reloading plugin Increment Selection.IncrementSelection
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin AutoHotkey.ahkexec
reloading plugin AutoHotkey.ahkgotodocs
reloading plugin AutoHotkey.ahkpopuphelp
reloading plugin AutoHotkey.utils
reloading plugin Deviot.DeviotStarter
plugins loaded
Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 162, in on_api_ready
    m.plugin_loaded()
  File "C:\Users\NHopkins\AppData\Roaming\Sublime Text 3\Packages\Deviot\DeviotStarter.py", line 61, in plugin_loaded
    Menu().createLibraryImportMenu()
  File "C:\Users\NHopkins\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs\Menu.py", line 164, in createLibraryImportMenu
    caption = json['name']
KeyError: 'name'
generating syntax summary
Package Control: No updated packages
Unable to parse command:

gepd, 2016-03-31:
I've release the version 1.1.6 of Deviot with a fix that should solve your issue, please make sure you have that version and let me know if it's working for you.

smartsystemdesign, 2016-03-31:
I was unable to upgrade to 1.1.6.  I also tried uninstalling the Deviot package and installing 1.1.6 via package control and the SublimeText console said:
```Package Control: Error downloading package. HTTP error 404 downloading https://codeload.github.com/gepd/Deviot/zip/v1.1.6.```

I also tried to load https://codeload.github.com/gepd/Deviot/zip/v1.1.6 in a web browser and got "Not found".

gepd, 2016-03-31:
Sorry about that, I had introduced a bug who doesn't allowed to install/update Deviot, you should now be able to do by package control without problems

smartsystemdesign, 2016-03-31:
Works like a charm now!  Thank you.

closed 2016-03-31 by smartsystemdesign