gepd/Deviot #13: Cannot select enviroment

peqc1300, 2016-03-30:
Hello,

As the tittle says I cannot select the environment. I have the board selected in the Select Board option but the Select Environment is grayed out.

```
Error says:
12:07:00 None environment selected
```

I have tried Teensy, Arduino Uno, Arduino Atmegaxxx and all of them give the same problem. 
Can you guide in how to start debugging this problem? What console utils I should use to debug it or if you know what the problem could be please let me know.

I have Arduino 1.6.7 and Teenyduino installed. I also have the platformio utility installed as well. Am missing packages to install?

I am using XUbuntu 14.04.

Thank you in advance and if you need more info please let me know!

closed 2016-03-30 by peqc1300

gepd, 2016-03-31:
Have you solved this issue?

peqc1300, 2016-03-31:
More than solved, is sorta  working.  iTt is sort of a glitch. If I click on another window not overlaping the Deviot menu button if I click directly (refocusing) in the Deviot button menu I am allowed to actually modify the environment. But if I focus anywhere on the sublime windows before clicking on the Deviot Seles Enviroment it is grayed out once again. Quite odd behavior. If you have any recommendations on how to debug this behavior I can certainly help and try and look for root cause. This is a great project so if there is anything I can do to help I available.

gepd, 2016-03-31:
I'll try yo explain you about that. 

Deviot allows you to select different environments based on your project/sketch. You can have a sketch with, for example, multiples arduino boards selected, and another one with multiples adafruit boards. So if you go to the tab with arduino boards, it will show you all the corresponding environments selected for that sketch and when you change to another tab, will update the list of available environments.

If you have an empty sketch, you should be able to select any environment previous selected from the board list.

It works like that because is the way of [PlatformIO](platformio.org) works (the core of Deviot) but any recommendation, suggestion is welcome.