gepd/Deviot #14: Autoscroll option for Serial Monitor

smartsystemdesign, 2016-03-31:
Can you add autoscrolling behavior to the serial monitor as in the Arduinio IDE serial monitor?  When checked, it automatically scrolls down to let you to see the newest data.

gepd, 2016-03-31:
I'll investigate if it is possible to do that in ST

smartsystemdesign, 2016-03-31:
Thank you.  This plug in is awesome.  Huge time saver to program in Sublime Text.

gepd, 2016-04-01:
Thanks @smartsystemdesign 

I've find a solution but It will available in the next relese, to add it manually please open the file

`libs > Messages.py`

Inside of the `MonitorView` class look for the method **println** and replace it for:

```
    def println(self, text):
        try:
            from .Preferences import Preferences
        except:
            from libs.Preferences import Preferences

        # Preferences to auto-scroll
        auto_scroll = Preferences().get('auto_scroll', True)

        self.view.set_read_only(False)
        pos = self.view.size()
        
        if python_version < 3:
            edit = self.view.begin_edit()           
            self.view.insert(edit, pos, text)
            self.view.end_edit(edit)
        else:
            if(auto_scroll):
                self.view.show(pos)
            self.view.run_command("append", {"characters": text})
        self.view.set_read_only(True)
```
make sure to copy the white spaces before `def println` to avoid sintaxis problems

You will be able to disable this option in the next release, in the main time it's activate by default

smartsystemdesign, 2016-04-01:
Thanks, a big improvement.  One more request, is it possibly to make it scroll after each line is printed instead of waiting for 30 lines and then scrolling?  That way, I can just look at the bottom of the page for current data.

For other users, libs can be found here:
C:\Users\UserName\AppData\Roaming\Sublime Text 3\Packages\Deviot\libs

gepd, 2016-04-01:
What O.S are you using? When I tested, it scrolled line by line, not after 30.

smartsystemdesign, 2016-04-01:
Windows 10 Pro 64 bit.

gepd, 2016-05-12:
I've released a new version of the plugin which includes the option `Auto-Scroll` unde the `Monitor Serial Menu` list

I've tested what you said about the 30 lines, it seems to be an issue of ST, but when I let the cursor in the last line, it seems to work better. I keep trying to look a better solution, but at least now I know what you mean.

closed 2016-05-12 by gepd