gepd/Deviot #16: None environment selected

Thijxx, 2016-04-04:
Get this error. No clue what environment it means.

gepd, 2016-04-04:
First you need to select one or more boards from the board list `Deviot Menu > Select Board(s)` after that, select the environment (board) you would like to work with `Menu > Select Environment`

You should do the same as in [this gif](https://github.com/gepd/Deviot#how-to-use-the-plugin)

Thijxx, 2016-04-04:
The 'Select Environment' is grayed out. I do have a board selected (Arduino Pro Mini)

gepd, 2016-04-04:
can you please copy and paste here the output of your console? `View > Show Console`

Thijxx, 2016-04-04:
These are the parts from the console. The last part repeats a couple of times.

`warning: mnemonic t not found in menu caption Tools`

```
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 493, in is_checked_
    ret = self.is_checked(**args)
TypeError: is_checked() got an unexpected keyword argument 'id_lang'
```

```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 497, in is_checked_
    ret = self.is_checked()
TypeError: is_checked() missing 1 required positional argument: 'lang_id'
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 493, in is_checked_
    ret = self.is_checked(**args)
TypeError: is_checked() got an unexpected keyword argument 'id_lang'
```

gepd, 2016-04-04:
What is you OS? are you using ST3?

Please copy the full output to debug the problem

Thijxx, 2016-04-04:
DPI scale: 1
startup, version: 3103 linux x64 channel: stable
executable: /opt/sublime_text/sublime_text
working dir: /
packages path: /home/thijs/.config/sublime-text-3/Packages
state path: /home/thijs/.config/sublime-text-3/Local
zip path: /opt/sublime_text/Packages
zip path: /home/thijs/.config/sublime-text-3/Installed Packages
ignored_packages: ["Vintage"]
scan: /home/thijs/.config/sublime-text-3/Packages/User/Deviot/penv/local/include has been seen before, skipping (using inode) previous path: /home/thijs/.config/sublime-text-3/Packages/User/Deviot/penv/include inode: 1460302
scan: /home/thijs/.config/sublime-text-3/Packages/User/Deviot/penv/local/bin has been seen before, skipping (using inode) previous path: /home/thijs/.config/sublime-text-3/Packages/User/Deviot/penv/bin inode: 1460233
scan: /home/thijs/.config/sublime-text-3/Packages/User/Deviot/penv/local/lib has been seen before, skipping (using inode) previous path: /home/thijs/.config/sublime-text-3/Packages/User/Deviot/penv/lib inode: 1460237
pre session restore time: 0.0908
warning: mnemonic t not found in menu caption Tools
startup time: 0.184211
first paint time: 0.202477
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.profile_syntax_definition
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Python.syntax_test_python
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.01-pygments
reloading plugin 0_package_control_loader.01-ssl-linux
Package Control: Linux SSL: successfully loaded _ssl module for libssl.so.1.0.0
reloading plugin 0_package_control_loader.02-bz2
reloading plugin 0_package_control_loader.50-pyyaml
reloading plugin AngularJS.AngularJS-sublime-package
reloading plugin AngularJS.jscompletions
reloading plugin AngularJS.message
reloading plugin AngularJS.sublime_commands
reloading plugin AngularJS.viewlocation
[... 193 lines cut ...]
TypeError: is_checked() missing 1 required positional argument: 'lang_id'
warning: mnemonic t not found in menu caption Tools
warning: mnemonic t not found in menu caption Tools
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 493, in is_checked_
    ret = self.is_checked(**args)
TypeError: is_checked() got an unexpected keyword argument 'id_lang'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 497, in is_checked_
    ret = self.is_checked()
TypeError: is_checked() missing 1 required positional argument: 'lang_id'
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 493, in is_checked_
    ret = self.is_checked(**args)
TypeError: is_checked() got an unexpected keyword argument 'id_lang'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 497, in is_checked_
    ret = self.is_checked()
TypeError: is_checked() missing 1 required positional argument: 'lang_id'

Thijxx, 2016-04-04:
Linux Mint 64-bit
ST3 3103

gepd, 2016-04-05:
I've released a new versión of Deviot with a fix for this problem, please check you're runing version 1.1.7 and tell me if the problem was solved.

Thijxx, 2016-04-05:
Ok, I think it's much better now. The Build now fails on a missing Lib. Not sure if Deviot should point to the Arduino IDE Lib folder in some way..

```
[ Deviot 1.1.7 ] attiny_sleeper.ino
08:16:43 Initializing the project | SUCCESS | it took 0.49s
08:16:43 Building the project | Processing...
/home/thijs/Repos/plantControl/arduino/attiny_sleeper/tmp_ino_to.cpp:3:24: fatal error: Manchester.h: No such file or directory
#include <manchester.h>
^
```

gepd, 2016-04-05:
At this moment deviot can't point to the Arduino library folder. Instead you have to copy or install the library.

* Open the library folder from `Deviot > Library Options > Open Libraries Folder` and copy your new library there.

* Install a library from the library manager `Deviot > Search Library`

Thijxx, 2016-04-05:
thanks, I will open an other issue for the libs if needed

closed 2016-04-05 by Thijxx