gepd/Deviot #17: ATtiny85 support

Thijxx, 2016-04-04:
Howdy, there is an ATMEL section under board but it does not show 'the usual suspects' like tiny45, 44, 84 and 85. Would you please add these as an enhancement?

gepd, 2016-04-04:
What are the `MCU` you are looking specifically? there is 4 `ATTINY85` in the board list 

Adafruit `Trinket 3V/8MHz - Trinket 5V/16MHz - Adafruit Gemma`
Digistump `Digispark (Default - 16 MHz) `

I can request the new boards to the platformio team (the core of deviot) so you can work without problem

Thijxx, 2016-04-04:
The `ATTINY85 20PU` which I use with `Arduino as ISP` with my Pro Mini, but I didn't look further than the Atmel menu to be honest..

gepd, 2016-05-12:
In the last version of platformio (2.9.1), is included a new support for the `ATTINY` boards, you can look for it, under the `Generic ATTiny` Menu.

If you don't have the last version of platformio go to `Deviot Menu > Options > Upgrade Platformio`

closed 2016-05-12 by gepd