gepd/Deviot #18: "No such file or directory" on lib file including

Eshkin-Kat, 2016-04-06:
I can't build any simplest project. Where does it look for library files? I have it in 3 directory:
- default Arduino "IDE" directories (c:\Program Files (x86)\Arduino\libraries\);
- C:\Users\%Username%\.platformio\lib;
- near the .ino file I'm trying to build.

There's file at c:\Users\%Username%\.platformio\lib\OneWire\OneWire.h
```
fatal error: OneWire.h: No such file or directory
#include <onewire.h>
```

Or I'm doing something wrong?

gepd, 2016-04-06:
All your libraries must to be in `c:\Users\%Username%\.platformio\lib` (`Deviot Menu > Library Options > Open Libraries Folder`), as you said you already did it, please provide me the code you are trying to compile to see what is the problem also:

What is your OS?
Are you using ST3?

Eshkin-Kat, 2016-04-06:
```
#include <OneWire.h>
OneWire ds(10);

void setup() {
  Serial.begin(9600); 
}

void loop() {
  byte data[2];
  ds.reset(); 
  ds.write(0xCC);
  ds.write(0x44);
  delay(750);
  ds.reset();
  ds.write(0xCC);
  ds.write(0xBE);
  data[0] = ds.read(); 
  data[1] = ds.read();
  int Temp = (data[1]<< 8)+data[0];
  int RealTemp = ds.read();
  Temp = Temp>>4;
  Serial.println(Temp);
  Serial.println(RealTemp);
}
```

Doesn't work with any code that contain including. Win 8.1. Sublime Text 3.

gepd, 2016-04-06:
I've tested your code and it compiles fine, please follow this steps:

1. Go to the menu `Deviot > Search Library`
2. Write `onewire` and select it from the list
3. Compile your code

The error `OneWire.h: No such file or directory` means the library isn't installed in the `Library Folder`

When you install a library manually, please always use the option menu: `Deviot > Library Options > Open Libraries Folder` to make sure you're installing it in the right place.

If you have problems with any other library, please let me know and we will try to solve it here.

Eshkin-Kat, 2016-04-06:
Thanks, it works now. Seems like Deviot requires some specific "installation", since regular copying library folders just doesn't work.

One last offtopic question. I don't know how and why but I lost build/upload/portmonitor buttons. I didn't touch any settings. Just opened SublimeText today and didn't see buttons. I can't find where I can turn it on.

gepd, 2016-04-06:
It's not necessary a specific/special installation, you can use same libraries as used in Arduino IDE, you only need to put it in the right folder.

Seems like I did some wrong changes in the last release, to bring it back, download [this file](https://www.dropbox.com/s/k5osontogl0ioaw/Main.sublime-menu?dl=0), go to `ST Menu > Preferences > Browse Packages` select the `Deviot/Settings-Default` folder and paste the file there.

Eshkin-Kat, 2016-04-07:
Ok, works. Thanks.

> you can use same libraries as used in Arduino IDE, you only need to put it in the right folder.

At first I opened Deviot Lib folder using Deviot menu. Then copied Arduino IDE libraries into that folder. For example, OneWire.h path remains the same after "installation" using SublimeText. But it just works now. `c:\Users\Анатолий\.platformio\lib\OneWire\OneWire.h`
And I have `c:\Users\Анатолий\.platformio\lib\FastLED\FastLED.h `which doesn't work for the same reason - not found. Works fine in Arduino IDE. I'm "Installing" it in Sublime Text with Deviot and it works now.
Could you tell me what exactly happens when I'm "installing" it? Does it download library from some repository and places it somewhere? Is path the same as Open Library Folders path?

gepd, 2016-04-07:
When you "install" a library from the manager, it only download the files from a repository.
All this is handled by [platformio](http://platformio.org) the core of Deviot. 

I'm thinking it may be because platformio adds a JSON file to organize the libraries. Can you provide me the links of the libraries you were using, so I can test it?

Eshkin-Kat, 2016-04-07:
I don't remember where I downloaded library. But maybe onewire was in Arduino IDE package. And I copied all default libs to Deviot library folder.
Next time I'll "install" all the libraries. Maybe it really will make platformio to understand that there is library that it can use.
Or you could make an option to read library folder and update that JSON file according. By the method it uses when I'm "installing" library from SublimeText.

gepd, 2016-04-09:
Today I've tested deviot with multiples libraries installed manually, I had no problem to using it. 

Can you please, add one of the libraries you had problem and after paste here the output of the ST console (`ST Menu > View > Show Console`) It will help me to solve the possible problem.

Thanks in advance.

zgoda, 2016-04-18:
Could this location be made configurable? Now I'm symlinking libraries to share them with Arduino IDE.

gepd, 2016-04-19:
At this moment isn't possible to change the location of the library folder, but there are plans to change that in the version 3 of platformio (the core of this plugin). So we have to wait until this version is published

closed 2016-04-19 by gepd