gepd/Deviot #20: Support for MKR1000

javorosas, 2016-04-14:
I just installed support for SAMD arduino boards, like the new MKR1000, from the boards manager of the Arduino IDE. I wonder how can I set Sublime/Deviot to work with the MKR1000

Thanks :)

gepd, 2016-04-15:
I've asked the state to support SAMD boards to the [platformio team](https://github.com/platformio/platformio/issues) (the core of deviot) we should have an answer soon

tabchas, 2016-06-01:
How can we use the development version of platformio? The issue apparently has been fixed as of today, but I couldn't find where we could use the develop version of platforimio in Deviot.

Here is link to SAMD issue at PlatformIO: https://github.com/platformio/platformio/issues/520#issuecomment-223125607

gepd, 2016-06-28:
Today platformio released a new version supporting the MKR1000 board, please update platformio from `Deviot menu > Options > Update platformio`

closed 2016-06-28 by gepd

javorosas, 2016-06-28:
Awesome!