gepd/Deviot #26: How to set board options ?

tablatronix, 2016-05-06:
How do we set board options ?
There is no board menu.

I tried modifying boards.txt to my preferences but it didn't work.

for example

esptool upload.speed ?

gepd, 2016-05-12:
I've released a new version with a new option `Open Build Config File`. I'll improve this in the future, but until know you can do this:

* Select the board you want to used
* Build/Compile the code
* Select `Open Build Config File`

After select that option you will see a file with some options follow [this instructions](http://docs.platformio.org/en/latest/platforms/espressif.html#custom-upload-speed) to modify the options as you want.

Please let me know if it solved your problem.

tablatronix, 2016-05-12:
Will try it tomorrow and probably test ota, wow platform ios docs are gorgeous.

tablatronix, 2016-05-13:
looks like you didn't update english lanuage, but ko instead ?

tablatronix, 2016-05-13:
Couple of questions

I am on mac, where should i set default_path to ?
Where should platformio.ini be saved?
Should it come up blank when editing ?

It looks like I have no existing platformio,ini file and that default path is set to current which came up as  my tmp build folder.

So I am not sure where to start.

gepd, 2016-05-13:
what do you mean, are you seeing all in korean?

I checked the english language file and it's fine https://github.com/gepd/Deviot/blob/master/Languages/en.lang

tablatronix, 2016-05-13:
Its all good, for some reason the lang file was not updated when I updated the plugin and was showing the tokens for the menu options, but now it is working.

Gonna close , seems to work.

closed 2016-05-13 by tablatronix