gepd/Deviot #27: Error Installing PlatformIO

alekseypb, 2016-05-06:
Hi!

I have Windows 7 x86, Sublime Text 3 build 3103.
Like to try you product.

Install python 2.7, after that install Deviot 1.1.7 by click Install from Sublime Text menu.
In start install, wait wait wait ... and got error Error Installing PlatformIO

I was restart Sublime, open console - again this error.

What i need to do?

> [ Deviot 1.1.7 ] Setup
> 00:58:12 PlatformIO is not installed
> Preparing to installing it, please be patient it may take a while.
> 00:58:12 Extracting files...
> 00:58:12 Installing PlatformIO...
> 00:58:27 Error Installing PlatformIO

alekseypb, 2016-05-07:
`DPI scale: 1
startup, version: 3103 windows x32 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/Bear/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/Bear/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/Bear/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Markdown", "Vintage"]
pre session restore time: 0.314549
startup time: 0.422549
first paint time: 0.449549
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.profile_syntax_definition
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Python.syntax_test_python
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.02-bz2
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Deviot.DeviotStarter
reloading plugin MarkdownEditing.bootstrap
reloading plugin MarkdownEditing.custom_find_under_expand
reloading plugin MarkdownEditing.distraction_free_mode
reloading plugin MarkdownEditing.footnotes
reloading plugin MarkdownEditing.gather_missing_links
reloading plugin MarkdownEditing.insert_references
reloading plugin MarkdownEditing.jumpToMarker
reloading plugin MarkdownEditing.list_markdown_references
reloading plugin MarkdownEditing.numbered_list
[... 118 lines cut ...]
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\requests\sessions.py", line 608, in send
    r.content
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\requests\models.py", line 737, in content
    self._content = bytes().join(self.iter_content(CONTENT_CHUNK_SIZE)) or bytes()
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\requests\models.py", line 660, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\requests\packages\urllib3\response.py", line 344, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\requests\packages\urllib3\response.py", line 301, in read
    data = self._fp.read(amt)
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 54, in read
    self.__callback(self.__buf.getvalue())
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\cachecontrol\controller.py", line 297, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\download.py", line 280, in set
    return super(SafeFileCache, self).set(*args, **kwargs)
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\cachecontrol\caches\file_cache.py", line 99, in set
    with self.lock_class(name) as lock:
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\lockfile\mkdirlockfile.py", line 19, in __init__
    LockBase.__init__(self, path, threaded, timeout)
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\site-packages\pip\_vendor\lockfile\__init__.py", line 242, in __init__
    hash(self.path)))
  File "c:\users\bear\appdata\roaming\sublim~1\packages\user\deviot\penv\lib\ntpath.py", line 85, in join
    result_path = result_path + p_path
UnicodeDecodeError: 'ascii' codec can't decode byte 0xcf in position 5: ordinal not in range(128)`

leglars, 2016-05-08:
yes, I meet the same issue, and have no idea what I need to do to fix it. Is there any possible that my default python is 3.4 not 2.7?

alekseypb, 2016-05-08:
i delete 2.7 now, install 3.4
run sublime and get this error
popup windows Deviot need python 2.7, please download and install it

leglars, 2016-05-08:
I think the problem could be there. I go to Plateform IO and download their Atom based IDE, and it gives me a warning that python3 don't be supported.

gepd, 2016-05-12:
Thanks for the report, as @leglars said the problem was you had installed python 3. I've pushed a new release, it will let you know when the python version is incompatible

closed 2016-05-12 by gepd