gepd/Deviot #28: Can't compile my code using Deviot, it is working from Arduino IDE.

nugrahawahyu, 2016-05-11:
[ Deviot 1.1.7 ] Fugue.ino
20:21:17 Initializing the project | SUCCESS | it took 1.73s
20:21:18 Building the project | Processing...
In file included from C:\Users\Nugraha\OneDrive\Tugas Akhir Deyen\Code\Fugue\tmp_ino_to.cpp:1:0:
.pioenvs\d1\StandardCplusplus_ID572/cstdlib:60:18: error: expected unqualified-id before 'long'
inline long abs(long i){
^
#define abs(x) ((x)>0?(x):-(x))
^
.pioenvs\d1\StandardCplusplus_ID572/cstdlib:60:18: error: expected ')' before 'long'
inline long abs(long i){
^
#define abs(x) ((x)>0?(x):-(x))
^
.pioenvs\d1\StandardCplusplus_ID572/cstdlib:60:18: error: expected ')' before 'long'
inline long abs(long i){
^
#define abs(x) ((x)>0?(x):-(x))
^
In file included from C:\Users\Nugraha\OneDrive\Tugas Akhir Deyen\Code\Fugue\vendor/nugraha/tugasAkhirWahyu/src/Nugraha/Header/Nugraha.h:38:0,
C:\Users\Nugraha\OneDrive\Tugas Akhir Deyen\Code\Fugue\vendor/nugraha/tugasAkhirWahyu/src/Nugraha/Header/../Devices/Device.h:34:49: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
debug::println("berhasil dihidupkan");
^
C:\Users\Nugraha\OneDrive\Tugas Akhir Deyen\Code\Fugue\vendor/nugraha/tugasAkhirWahyu/src/Nugraha/Header/../Devices/Device.h:46:48: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
debug::println("berhasil dimatikan");
^
C:/Users/Nugraha/OneDrive/Tugas Akhir Deyen/Code/Fugue/Fugue.ino:19:6: error: 'void Main::Controller::setup()' cannot be overloaded
{
^
C:/Users/Nugraha/OneDrive/Tugas Akhir Deyen/Code/Fugue/Fugue.ino:18:10: error: with 'void Main::Controller::setup()'
void setup()
^
C:/Users/Nugraha/OneDrive/Tugas Akhir Deyen/Code/Fugue/Fugue.ino:25:10: error: 'void Main::Controller::loop()' cannot be overloaded
{
^
C:/Users/Nugraha/OneDrive/Tugas Akhir Deyen/Code/Fugue/Fugue.ino:20:10: error: with 'void Main::Controller::loop()'
debug::isdebugmode = true;
^
20:21:29 ERROR | it took 10.15s

gepd, 2016-05-12:
hi @nugrahawahyu try to upload the code in any service so I can make the tests, and look for a solution, without that is not much what I can do

nugrahawahyu, 2016-05-13:
Code added,.

gepd, 2016-05-14:
I have compiled the code without problems, it may be because the last update of platformio, please check if there is a new version available to you in `Deviot Menu > Options > Update Platformio`

After that try to compile again

closed 2016-06-02 by gepd