gepd/Deviot #31: Update to development PlatformIO version

tabchas, 2016-06-02:
How can we use the development version of PlatformIO? There is a SAMD uploading issue which has  been fixed as of today, but is only available on the latest dev version of PlatformIO. 

Atom has an option for PlatformIO to update to latest (even non-stable versions), does Deviot have something similar?

Here is link to SAMD issue at PlatformIO: platformio/platformio#520 (comment)

gepd, 2016-06-02:
A fast solution I can bring you is:

- Download [this branch](https://github.com/gepd/Deviot/archive/feature/pio_develop.zip)
- Remove the previous version of deviot; `ST Menu > Preferences > Browse Packages` Remove the `Deviot` folder
- Remove the deviot user preferences folder;  `ST Menu > Preferences > Browse Packages` go to `User` folder and remove `Deviot`
- Restart ST

After that, the`platformio dev` version should start to be installed
I'll improve this feature in the next release.

tabchas, 2016-06-02:
Sweet! A suggestion is to have a little check mark option in the settings.

For reference, here is how it is currently implemented in the Atom IDE for PlatformIO.

[image]

tabchas, 2016-06-02:
@gepd Hmm, are you sure that its pulling from the develop branch of PlatformIO? Where is that specified in the code?

gepd, 2016-06-02:
I'm sure: https://github.com/gepd/Deviot/blob/feature/pio_develop/libs/Install.py#L188

tabchas, 2016-06-06:
@gepd Sweet, looks like its working! Thanks for the quick fix. Will be a good addition for a future release 👍 .

closed 2016-06-11 by gepd