gepd/Deviot #32: Serial monitor output in console window

tabchas, 2016-06-02:
Is there a way to output the serial monitor in the console window? I think this would be much better so you can still see the code and the Serial monitor output at the same time.

gepd, 2016-06-02:
I don't think you can output the serial monitor in the windows console, but you can split the ST view in two.

There are two ways to do this
- Go to `View > Layout > Columns: 2` or `Rows: 2`
- Using the shortcuts; `Alt + Shift + 2` (Columns) `Alt + Shift + 8` (Rows)

tabchas, 2016-06-02:
Hmm, I meant in the Sublime Text console on the bottom. You currently use it during build/upload to output details during the process.

Can you print serial port data there? That's how its implemented in the Atom IDE, and I'd prefer it that way. Not sure what other people think.

gepd, 2016-08-18:
This feature was added in the las version of Deviot `1.2.0`, you can choose it from `Deviot Menu > Monitor Serial > Output in Deviot Console`

closed 2016-09-02 by gepd