gepd/Deviot #33: Error: Unknown environment names 'due'. Valid names are 'dueUSB'

guilhermeaugustos, 2016-06-03:
I can compile and upload for Due using progamming port at first. But when next I modify something and try to reupload this erros apears.

gepd, 2016-06-03:
Please go to `Deviot Menu > Options > Update Platformio` if you have an old version of platformio it will be updated, if it doesn't solve the problem, please let me here the code are you using to try other solution.

guilhermeaugustos, 2016-06-04:
Really, my Platformio it was not updated. Apparently everything is working fine, thank you for your help.

closed 2016-06-04 by guilhermeaugustos

thomsna, 2016-09-19:
I get the same type of error when selecting Arduino Zero (USB Native Port):

Error: Unknown environment names 'zeroUSB'. Valid names are 'mzeroproUSB, teensy31, pro16MHzatmega328, micro, mzeroUSB, uno'

thomsna, 2016-09-19:
I don't have the error anymore after I deselected "Keep Temp Files" in Deviot Menu > Options

ahart2869, 2022-11-28:
I get the below error when attempting to compile what am I doing wrong.
Error: Unknown environment names 'STM32F103RC_creality'. Valid names are 'include_tree'