gepd/Deviot #34: deviot do not appear on the st3 menu

gnkarn, 2016-06-25:
after installing all , and receiving successful installation comment , deviot menu do not appear on the st3 head menu.

gepd, 2016-06-25:
What S.O are you using? can you provide me the output of you console? (`View > Show console`)

gnkarn, 2016-06-25:
thanks for your response,.

osx el capitan

startup, version: 3114 osx x64 channel: stable
executable: /Applications/Sublime Text.app/Contents/MacOS/Sublime Text
working dir: /
packages path: /Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages
state path: /Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Local
zip path: /Applications/Sublime Text.app/Contents/MacOS/Packages
zip path: /Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Installed Packages
ignored_packages: []
pre session restore time: 0.946796
startup time: 1.01607
first paint time: 1.02248
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Vintage.vintage
reloading plugin Vintage.vintage_commands
reloading plugin Vintage.vintage_motions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.02-bz2
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Deviot.DeviotStarter
plugins loaded
Traceback (most recent call last):
  File "/Applications/Sublime Text.app/Contents/MacOS/sublime_plugin.py", line 162, in on_api_ready
    m.plugin_loaded()
  File "/Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages/Deviot/DeviotStarter.py", line 58, in plugin_loaded
    Tools.createCompletions()
  File "/Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Tools.py", line 406, in createCompletions
    keywords = getKeywords()
  File "/Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Tools.py", line 384, in getKeywords
    keywords_dirs = Paths.getLibraryFolders()
  File "/Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Paths.py", line 400, in getLibraryFolders
    pio_lib_path = getPioLibrary()
  File "/Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Paths.py", line 163, in getPioLibrary
    raise exc
  File "/Users/Gus-nuevo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Paths.py", line 160, in getPioLibrary
    os.makedirs(pio_lib)
  File "./python3.3/os.py", line 251, in makedirs
  File "./python3.3/os.py", line 262, in makedirs
PermissionError: [Errno 13] Permission denied: '/Users/Gus-nuevo/.platformio'
Package Control: No updated packages

gepd, 2016-06-28:
I'm sorry for the late answer...

It's a permission related issue , I've found two possible solutions:

1. Add `os.chmod(pio_lib, 0777)` in line 161 (after `os.makedirs(pio_lib)`) in the file `libs/Paths.py`  
Delete the `Deviot` folder in `Packages/User/` and restart Deviot

2. Try running this command in the terminal: `chmod 777 /Users/Gus-nuevo/`  
Delete the `Deviot` folder in `Packages/User/` and restart Deviot

Are you using an admin account?

Let me know if one of this solution helped you

gnkarn, 2016-06-28:
Thank you, ! I will try this Friday and gave feedback then.

-------- Mensaje original --------

Fecha: 28/06/2016  1:11  (GMT-03:00) 
A: gepd/Deviot <Deviot@noreply.github.com> 
CC: gus <gnkarn@gmail.com>, Author <author@noreply.github.com>

gnkarn, 2016-07-02:
thanks a lot  ! ,

it works fine now ,

gustavo

gepd, 2016-07-06:
which option worked finally?

closed 2016-07-11 by gepd