gepd/Deviot #35: updating cores ?

tablatronix, 2016-06-28:
esp8266 core was updated to 2.3.0.
How do i go about updating it in deviot/stino/platformio

Is there any useful cli support for this in either of these platforms ?

gepd, 2016-06-28:
Yes, you can update platformio from `Deviot menu > Options > Update platformio`
I'll improve this feature in the next release.

closed 2016-06-29 by gepd

tablatronix, 2016-07-04:
when i run this however 
it does not say anything, just stops.

```
[ Deviot 1.1.9 ]
Checking for PlatformIO updates...
```

gepd, 2016-07-04:
Please update deviot from the package control (`ctrl+shift+p` > `Package Control: Upgrade Package`) it was a bug, I've fixed in that version (`v1.1.10`)

tablatronix, 2016-07-04:
working now, package manager was not picking up 1.1.10, must have been cached, I had to restart sublime text.

Thanks.