gepd/Deviot #37: Deviot does not appear

avrosmalen, 2016-07-02:
Deviot does not appear on the menu and keeps asking for Python 2.7 even though Python 2.7 was installed.

Using latest version of Sublime (build 3114), installed Deviot through package control and installed the latest version of Python 2.7. Python 3 is not installed.

gepd, 2016-07-03:
Are you using windows? which version?

avrosmalen, 2016-07-03:
Windows 10

emogg, 2016-07-03:
Same issue with Linux (Archlinux) Python 2.7 is installed

gepd, 2016-07-03:
I've just tried a new installation in a virtual machine and it works fine to me

can you paste the output of your console?

avrosmalen, 2016-07-03:
What do you mean with adding Python to the system path?

Console output:
DPI scale: 1
startup, version: 3114 windows x64 channel: stable
executable: /C/Program Files/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files/Sublime Text 3
packages path: /C/Users/username/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/username/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files/Sublime Text 3/Packages
zip path: /C/Users/username/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.127587
startup time: 0.156587
first paint time: 0.158587
first paint time: 0.158587
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_file_settings
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin 0_package_control_loader.00-package_control
reloading plugin 0_package_control_loader.02-bz2
reloading plugin Package Control.1_reloader
reloading plugin Package Control.2_bootstrap
reloading plugin Package Control.Package Control
reloading plugin Deviot.DeviotStarter
plugins loaded
'python' is not recognized as an internal or external command,
operable program or batch file.

Package Control: No updated packages

gepd, 2016-07-03:
Yes, your problem seems to be related with the system path. Look [this](https://github.com/gepd/Deviot/blob/master/Docs/images/win_python.gif) gif

avrosmalen, 2016-07-04:
Updated the installation, checked the option but Deviot is still invisible and Sublime is still telling me to install Python...

gepd, 2016-07-04:
Did you restarted you O.S?

this line in your console: `'python' is not recognized as an internal or external command` means python was not found in your system

if you open a window console and type `python`, it runs python?

avrosmalen, 2016-07-07:
Awesome, that did the trick. I had to restart the whole PC and not just Sublime Text. Thanks.

closed 2016-07-07 by avrosmalen