gepd/Deviot #38: Portable Sublime and Python

eusebiotebar, 2016-07-06:
Hi
MY SETUP:
I'm running a portable version of Python 2.7.6.1 (http://portablepython.com/wiki/PortablePython2.7.6.1/ ) on a USB stick, installed under: "E:\MyTools\Python\App"

Also, I'm running a portable version of Sublime Text 3 Build 3059 (http://c758482.r82.cf2.rackcdn.com/Sublime%20Text%20Build%203059.zip ) on the same USB stick, installed under: "E:\MyTools\Sublime Text 3\"

Edited the "E:\MyTools\Sublime Text 3\Data\Packages\Deviot\Settings-Default\Deviot Preferences.sublime-settings" with this line:
"env_path": "E:\\MyTools\\Python\\App","E:\\MyTools\\Python\\App\\Scripts"

Results:
When Sublime stars, Devion ask again for instal python 2.7

Any idea for setting?

gepd, 2016-07-07:
HI @eusebiotebar 

I've added a fix for this issue. Please overwrite the file `Packages/Deviot/libs/Install.py` for [this](https://www.dropbox.com/s/i9wq5gmst1ox2a9/Install.py?dl=0) one

And sets the env_path var in this way: 
`"env_path": "E:\\MyTools\\Python\\App;E:\\MyTools\\Python\\App\\Scripts"`

It should fix the problem

eusebiotebar, 2016-07-07:
Same result

Output from console
================
C:\Python27\;C:\Python27\Scripts;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem
================

gepd, 2016-07-07:
I'm Sorry I just realized now you are editing the wrong file, the `Deviot Preferences.sublime-settings` is located in: `E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot`

The file should looks like this:

```
{
    "env_path": "E:\\MyTools\\Python\\App;E:\\MyTools\\Python\\App\\Scripts", 
    "id_lang": "en"
}
```

eusebiotebar, 2016-07-07:
It works.... more or less :100:.. I mean that now it tries to install PlatformIO but after a while, the installation says 

`Error Creating the virtualenv` 

I don't think so it is a problem from Deviot but a PlatformIO issue. Probably it tries to install it over the hard disk (which I don't have the proper rights) instead the 'portable' disk.

gepd, 2016-07-07:
It may be a permission issue, can you see if the console shows any aditional information? (and paste here)

Can't you run ST as administrator?

eusebiotebar, 2016-07-07:
I can't run ST as admin in this computer. I will try this afternoon at home.

The only output from consoly is this:
`[ Deviot 1.1.10 ] Instalación`
`11:36:39 PlatformIO no está instalado`
`Preparándose para instalarlo. Por favor sea paciente esto podría demorar un poco`
`11:36:39 Extrayendo archivos...`
`11:36:39 Instalando PlatformIO...`
`11:37:07 Error Creating the virtualenv`

gepd, 2016-07-07:
The ST console can show extra information `ST Menu View > Show Console` it's different from deviot console

eusebiotebar, 2016-07-07:
My mistake, I was thinking that the output was from ST console.

This is the ST console Output. 

```
================
""E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\pio"" no se reconoce como un comando interno o externo,
programa o archivo por lotes ejecutable.

Package Control: Skipping automatic upgrade, last run at 2016-07-07 11:36:49, next run at 2016-07-07 12:36:49 or after
New python executable in E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe
ERROR: The executable "E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe" could not be run: [Error 5] Acceso denegado
```

It seems that I should have admin rights, but I can execute from external the command
`"E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe" `

eusebiotebar, 2016-07-11:
The same result and console output in a personal home computer.

gepd, 2016-07-11:
It's possible we can schedule a teamviewer meeting so I can debug this problem? you can write me to get@outlook.com

eusebiotebar, 2016-07-14:
The error message says:
`New python executable in E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe`
`ERROR: The executable "E:\MyTools\Sublime Text 3\Data\Packages\User\Deviot\penv\Scripts\python.exe"`

So in some part you do copy for `python.exe` in that folder, why?

gepd, 2016-07-15:
To install platformio, the core of deviot, the plugin creates a virtual environment. The virtual environment install python inside of that folder. All that, is to install deviot and the dependencies automatically.

asrimane, 2016-07-30:
Hi, 
I have the same error:

`New python executable in C:\Users\***\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\python.exe`
`ERROR: The executable "C:\Users\***\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\python.exe" could not be run: [Error 5] Accès refus`

what is the script that runs the executable `Deviot\penv\Scripts\python.exe` ?
I want to try to copy it somewhere else.
The problem does not come from the apparently antivirus

@+

gepd, 2016-07-30:
The whole plugin uses that executable, I've tried to reproduce this issue but I can't, are you using Windows 10 too?

One command can be

`C:\Users\***\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\python --version`
or
`C:\Users\***\AppData\Roaming\Sublime Text 3\Packages\User\Deviot\penv\Scripts\pio --version`

If you want we can check this by Teamviewer
you can contactme at gepd@outlook.com

asrimane, 2016-07-31:
No i have Windows 8.1 64bit

gepd, 2016-08-01:
I'm sorry, as I can't replicate this issue isn't much I can do by now

gepd, 2016-09-27:
Re open this issue if you have new information or if you are willing to arrange a meeting with me in teamviewer or something similar.

closed 2016-09-27 by gepd