gepd/Deviot #39: Error mal indicado

Surbyte, 2016-07-07:
Un simple error como poner un float con coma (,) y no (.) punto decimal genera esto

[image]
La parte de código que muestra el error es
`void loop() {
  float valor = analogRead(A0) * 0,00782;  //0,00782 = 5/1023
  Serial.println(valor);

  delay(1000);  
}`

El error mostrado

`[ Deviot 1.1.10 ] Borrar_3.ino
15:13:24 Compilando el proyecto | Procesando...
[07/07/16 15:13:24] Processing uno (platform: atmelavr, board: uno, framework: arduino)

avr-g++ -o .pioenvs\uno\src\tmp_ino_to.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021001 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs\uno\FrameworkArduino -I.pioenvs\uno\FrameworkArduinoVariant -ID:\Dropbox\Arduino\Foro\Borrar\Borrar_3 D:\Dropbox\Arduino\Foro\Borrar\Borrar_3\tmp_ino_to.cpp
D:/Dropbox/Arduino/Foro/Borrar/Borrar_3/Borrar_3.ino:28:36: error: invalid digit "8" in octal constant
D:/Dropbox/Arduino/Foro/Borrar/Borrar_3/Borrar_3.ino: In function 'void loop()':
D:/Dropbox/Arduino/Foro/Borrar/Borrar_3/Borrar_3.ino:28:36: error: expected unqualified-id before numeric constant
avr-gcc -o .pioenvs\uno\FrameworkArduino\WInterrupts.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021001 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs\uno\FrameworkArduino -I.pioenvs\uno\FrameworkArduinoVariant .pioenvs\uno\FrameworkArduino\WInterrupts.c
avr-g++ -o .pioenvs\uno\FrameworkArduino\WMath.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021001 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs\uno\FrameworkArduino -I.pioenvs\uno\FrameworkArduinoVariant .pioenvs\uno\FrameworkArduino\WMath.cpp
scons: *** [.pioenvs\uno\src\tmp_ino_to.o] Error 1
avr-g++ -o .pioenvs\uno\FrameworkArduino\WString.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021001 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs\uno\FrameworkArduino -I.pioenvs\uno\FrameworkArduinoVariant .pioenvs\uno\FrameworkArduino\WString.cpp
 [ ERROR ] Took 1.21 seconds `

gepd, 2016-07-08:
Estuve mirando y el problema está en el código que muestra el error de forma reducida. Estoy trabajando en la versión 1.2, incluiré la corrección ahí, por ahora la solución sería activar el log completo de compilación `Deviot Menu > Salida detallada al Compilar`

gepd, 2016-08-18:
Solucionado en v1.2.0

closed 2016-08-18 by gepd