gepd/Deviot #40: Programming attiny via USBtinyISP?

nvasilakes, 2016-07-08:
I was able to successfully program my attiny85 with the normal arduino ide by installing attiny boards and selecting USBtinyISP as the programmer.

Is there a way to do this in Deviot? I've tried using GenericATTiny85 but it doesn't do anything. Reading up on platform.io it looks like it needs to be using the program command instead of the upload command.

http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

I've tried adding "upload_protocol = usbtiny" to the build config file in Deviot but it wasn't successful, probably because it's using the upload command still?

I'm rather new at this and would appreciate any help!

gepd, 2016-07-08:
Probably it's because the upload command. I'm currently working in that feature for v1.2, I'll try to upload a fast fix for you. I hope to back soon :)

gepd, 2016-07-26:
Hey @nvasilakes  version 1.2 is almost ready and I've implemented this feature.

Please, download the files from [here](https://github.com/gepd/Deviot/archive/feature/v1.2.0.zip) replace the `Deviot` folder in `Packages` folder

Note that in the status bar shows `v1.1.10` when the final version is released, it will update automatically to `v1.2.0` with the possible fixes/changes.

I'll appreciate any feedback on this feature

nvasilakes, 2016-07-28:
My serial ports all disappear when I replace the Deviot folder...am I doing something wrong?

nvasilakes, 2016-07-28:
Also, what exactly did you change? Should I be able to add the upload_protocol manually into the build config file and it works now?

gepd, 2016-07-28:
Serial ports now uses the `quick search` ST feature to show the available ports. You need to click on the menu. It's because now Deviot can autodiscover mDNS services/devices.

To use the programmer feature you have to choose any programmer option in the menu
[image]

It will add automatically the `upload_protocol` line(s) in the config file. But if you need to add any extra flag/configuration to the platformio.ini file you should use `Deviot Menu > Open Build Config File`

nvasilakes, 2016-07-28:
Should I be completely replacing Packages/Deviot with the contents of Deviot-feature-v1.2.0.zip? When I do that I lose the items in my Deviot menu. When I select the drop down there are only 3 items, all blank. One expands into examples, one into language, and one into select environment

I also tried simply overwriting any files of the same name with the new ones. In that case I still have my menu but there is no Programmer option.

gepd, 2016-07-29:
What is your OS?

nvasilakes, 2016-07-29:
Windows 10

gepd, 2016-07-29:
Remove the folder `Deviot` in `Packages\User\` and restart ST, the plugin will be reinstalled. It should solve any problem, if you continue with the same issue paste the console output here

nvasilakes, 2016-07-29:
Thanks, it looks like it has installed correctly. I am, however, missing the shortcut buttons that normally appear to the right of the Deviot menu.

Unfortunately I haven't been able to get it to work. I'm getting the following error:

```
20:25:04 Uploading firmware | 
avrdude: stk500_recv(): El programador no responde
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03

avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xe0

avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xe0

avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xe0

avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00

avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xe0

avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xe0

avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xe0

avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00

avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xe0
20:25:18 ERROR | it took 14.25s
```

I am able to successfully upload the hex file using avrdude with the following command:

`avrdude -c avrisp -p m328p -P COM5 -b 19200 -U flash:w:firmware.hex`

Is there some way to find out exactly what command is being used with avrdude? I thought it might be the baud rate but I have it set correctly in Deviot and I also tried editing the build config file by adding `upload_speed = 19200` which had no effect.

gepd, 2016-07-29:
Go to `Deviot Menu > Options > Full Verbose Output When Build` that way you will get the full output including the parameters passed to `avrdude`

nvasilakes, 2016-07-29:
Great, verbose output was very helpful. The reason it's not working is indeed the baud rate, as I suspected. I've manually edited the build config file platformio.ini and saved it, but as soon as I choose upload the platformio.ini is overwritten & my changes are removed and my flag isn't sent. 

I have tried the following variations with no success:

```
[env:uno]
platform = atmelavr
framework = arduino
board = uno
upload_protocol = stk500v1
upload_flags = -P$UPLOAD_PORT -b19200
upload_port = COM5
```

```
[env:uno]
platform = atmelavr
framework = arduino
board = uno
upload_protocol = stk500v1
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_port = COM5
upload_speed = 19200
```

```
[env:uno]
platform = atmelavr
framework = arduino
board = uno
upload_protocol = stk500v1
upload_flags = -b$UPLOAD_SPEED
upload_speed = 19200
upload_flags = -P$UPLOAD_PORT
upload_port = COM5
```

```
[env:uno]
platform = atmelavr
framework = arduino
board = uno
upload_protocol = stk500v1
upload_flags = -P$UPLOAD_PORT
upload_port = COM5
upload_flags = -b$UPLOAD_SPEED
upload_speed = 19200
```

But as soon as I hit upload, it instantly changes back to 

```
platform = atmelavr
framework = arduino
board = uno
upload_protocol = stk500v1
upload_flags = -P$UPLOAD_PORT
upload_port = COM5
```

So no matter what I do the avrdude command line is still being sent as `avrdude" -v -p atmega328p -C "C:\Users\nicho\.platformio\packages\tool-avrdude\avrdude.conf" -c stk500v1 -PCOM5 -U flash:w:.pioenvs\uno\firmware.hex:i` with no baud rate flag.

gepd, 2016-07-29:
I pushed a change to fix this issue, replace your current `PlatformioCLI.py` for [this one](https://github.com/gepd/Deviot/blob/feature/v1.2.0/libs/PlatformioCLI.py) in `Packages\Deviot\Lib`

nvasilakes, 2016-07-29:
That did it - works great now!

Question: did you remove the quick buttons for build and upload that used to appear to the right of the Deviot menu or am I just missing an option somewhere?

gepd, 2016-07-29:
I did it, the problem was it only fully worked on Windows 10, I'm looking for a new solution, but if you want to recover it save [this file](https://github.com/gepd/Deviot/blob/master/Preset/Main.sublime-menu.windows) as `Main.sublime-menu` and save in `Packages/Deviot/Preset`

I've also added new shortcuts, with the main functions of the plugin, in the contextual menu. At least it seems to be a little more comfortable for me

nvasilakes, 2016-07-29:
Excellent, thanks!

gepd, 2016-07-30:
I'll close this issue, if you have any other problem related to this you can re open it, or open a new one.

closed 2016-07-30 by gepd