gepd/Deviot #41: I can't seem to point Deviot to a user defined library location

avrosmalen, 2016-07-12:
Right now Deviot seems to stick to keeping its libraries at one specific location which cannot be changed. This is highly unpractical as this gives you files to maintain at multiple locations. Did I overlook some setting or is this simply a bug that needs to be fixed?

gepd, 2016-07-15:
It isn't a bug, at this moment, is not possible to change the library folder because the platformio limitations (the core of deviot). But the platformio team is already working in the version 3 which will give the option to specify new libraries in a custom folder

avrosmalen, 2016-07-16:
Thanks! I will just wait a bit then then before using Deviot.

closed 2016-07-16 by avrosmalen

stormbytes, 2016-09-10:
Dunno if this would work, but (I'm on a mac) you could technically try deleting ~/.platformio lib and create a sim-link (and call it "lib") in ~/.platformio/ and pint it to a folder of your choosing -- or even the Arduino IDE library folder. Its a hack but hey, it would do the trick :) (if it works)

gepd, 2016-09-10:
the new version of platformio is released now, it supports new paths for libraries... 

@stormbytes where Arduino IDE create the Arduino folder to add user libraries in OSX? in Windows it's in `username/Documents/Arduino/Libraries` but I haven't a mac close to check that

stormbytes, 2016-09-10:
I don't know how to create sim-links on Windows (I think they are called "shortcuts" ?) Presumably you'd simply create a "shortcut" to the #/Documents/Arduino/Libraries folder and call it "libs" so the name matches, then drag that folder/shortcut into the platformio folder (deleting platformio/lib first). 

If the new version of Platformio allows you to set the lib folder wherever you like there's really no need for you to do any of this :)

gepd, 2016-09-10:
Yes, it's not need, my question it's because I'll add a feature to recognize automatically the libraries that usually are in the arduino folder (if you used to use the IDE)

stormbytes, 2016-09-10:
I don't personally use the IDE (where I can help it!) which is why I'm so
glad to have found Deviot.

I think that autodetection of existing IDE libs would be a terrific feature
addition, but I would also add an option to disable that in case some
people would like to keep libraries project-based.

gepd, 2016-09-10:
Don't worry, as always, it will be optional :smile:

stormbytes, 2016-09-10:
Good stuff! :)