gepd/Deviot #43: Feature console auto scroll

tablatronix, 2016-07-15:
It would be nice to implement autoscroll on the console, without stealing focus from editor.

I notice if you focus console, hit end , and the blur, it stays scrolled.

Might be a useful workaround or implementation available,.

tablatronix, 2016-07-15:
Also if you try to use the search bar the console disappears, but if it pops back up if there is Activity with the same size.

gepd, 2016-07-16:
Yes, I still have to know how to solve this issue. I got a closer solution in the past, but it needs a little more of work yet

tablatronix, 2016-07-16:
great it seems st must have some autoscroll built in if cursor is at the end of a buffer or something

tablatronix, 2016-07-16:
It looks like the build system consoles are similar, would it be possible to hook into the actual build menu, or is that how it already works ?

tablatronix, 2016-07-16:
There is also an option to reshow build results in the menu already

gepd, 2016-07-16:
Yes, that is the solution I found, it's not hard to change. I've done some tests and it seems to work. I'll let you know when it be ready, I still have to work in other things to version 1.2 but will be possible to test the new features from the branch I'm working on

tablatronix, 2016-07-16:
Oh ok ill check it out is it stable enough to qa?

gepd, 2016-07-17:
Not yet, I'll let you know when a usable version is available

gepd, 2016-07-26:
Hey @tablatronix version 1.2 is almost ready and you can try it now.

Download the files from [here](https://github.com/gepd/Deviot/archive/feature/v1.2.0.zip) replace the `Deviot` folder in `Packages` folder

Note that in the status bar shows `v1.1.10` when the final version is released, it will update automatically to `v1.2.0` with the possible fixes/changes. 

This version has many new features, use programmer to upload, autodiscover mDNS services, ota uploads in esp, and others things. If you try any of them and can give me any feedback, it will be appreciated.

tablatronix, 2016-07-29:
Looking good so far, how do you want dev bugs reported ?
I noticed my environment and serial ports menus are empty.

gepd, 2016-07-29:
Select Board/Environment and Serial Ports now works with quick panel, you have to click on the menu to select a board/port

Is that not working for you?

tablatronix, 2016-07-29:
I can click select board, and the list of boards shows up.
But my environment has no arrows or submenus.
and my serial ports also has no arrow or submenus

gepd, 2016-07-30:
I've fixed the bug, please remove the `Deviot` folder in `Packages/User/Deviot`, after restart ST it will reinstall the plugin

gepd, 2016-07-30:
I'll close this issue, for any other problem I don't mind if you open a new one

closed 2016-07-30 by gepd