gepd/Deviot #44: Console closes and cannot be reopened

tablatronix, 2016-07-15:
I do not see a way to bring back console.

hitting esc or bringing up sublime console, hides the deviot console.

tablatronix, 2016-07-15:
Also if you try to use the search bar the console disappears, but it pops back up if there is Activity with the same size. So searching is not possible when the console is busy.

Might have to consolidate all these, as they seem to be sublime limitations perhaps.

gepd, 2016-07-16:
The same solution I've found for https://github.com/gepd/Deviot/issues/42 will allow to reopen the console. The issue with the search bar sadly there are not much to do because is how ST works

gepd, 2016-07-30:
Same here: I'll close this issue, if you find any problem with this feature, please re-open it

closed 2016-07-30 by gepd