gepd/Deviot #46: 1.2 cant upload to board

tablatronix, 2016-07-30:
I can upload from arduino ide with no problem.

But I get cannot access serial port, I have never seen this issue before.

```
MethodWrapper(["upload"], [".pioenvs/huzzah/firmware.bin"])
Looking for upload port/disk...
Manually specified: /dev/cu.usbserial-a100vng2
"/Users/.../.platformio/packages/tool-esptool/esptool" -vv -cd nodemcu -cb 115200 -cp "/dev/cu.usbserial-a100vng2" -cf .pioenvs/huzzah/firmware.bin
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
setting board to nodemcu
setting baudrate from 115200 to 115200
setting port from /dev/tty.usbserial to /dev/cu.usbserial-a100vng2
espcomm_upload_file
espcomm_upload_mem
opening port /dev/cu.usbserial-a100vng2 at 115200
error: cannot access /dev/cu.usbserial-a100vng2

error: espcomm_open failed
error: espcomm_upload_mem failed
scons: *** [upload] Error 255
========================= [ ERROR ] Took 6.05 seconds =========================
09:00:48 ERROR | it took 6.35s
```

any ideas on `error: cannot access`
I can run serial monitor just fine, and arduino ide uploads just fine.

should i try esptool from cmd line and see what it does.

tablatronix, 2016-07-30:
Serial port is case sensitive as least on osx

`/dev/cu.usbserial-A100VNG2`
works, so it must be getting lowercased

tablatronix, 2016-07-30:
looks like it is saved correctly in prefs and serial.json, so it must be the esptool builder code

gepd, 2016-07-30:
I've pushed a change in the branch, please, rewrite the deviot folder and tell me if it works now

tablatronix, 2016-07-30:
works

closed 2016-07-30 by tablatronix