gepd/Deviot #49: 1.2.0 libraries not refreshing

tablatronix, 2016-08-03:
When I manually add libraries, they are not picked up, nor are examples.
If I revert to stable it works.

Maybe I am using the wrong lib location...
user/.platformio/lib

Is there a user lib folder?

tablatronix, 2016-08-03:
Wont refresh for removals either

gepd, 2016-08-04:
Yes, that is the folder. After copy the library, did you restarted Sublime?
I've tried that now with a random library, and it's working to me

I've to add the quick panel to avoid restar ST, I should do it this weekend

tablatronix, 2016-08-04:
Heh yeah I tried restarting over and over. Im on osx

gepd, 2016-08-04:
I'll test in osx to see what is the problem

tablatronix, 2016-08-04:
It could be something localized to me but I tested for a while removing and reinstalling libs and reverting to stable etc.

gepd, 2016-08-15:
I've pushed some changes, and now the libraries are using the quick panel, this issue shouldn't be a problem anymore. 

Reopens this issue if you found any related problem

closed 2016-08-15 by gepd