gepd/Deviot #51: feature platformio command line

tablatronix, 2016-08-03:
was just thinking it might be useful to run platform io commands in env from a cmd line entry.
( I do not think st3 has a robust console, wbond has something though )

tablatronix, 2016-08-03:
by console I mean like terminal in platformio ide, etc

gepd, 2016-08-05:
I'll check about this, I've this on my mind for a while, but I haven't had the time to check it. After release v1.2 I'll start to work on that

tablatronix, 2016-08-06:
https://github.com/chrissimpkins/glue
Thats the one I saw i have not tried it yet.

gepd, 2016-09-03:
I've been testing this, it seems very interesting I'll start to do some implementations in Deviot, thanks for pointing it!
It will be the first feature to the next mayor release!

tablatronix, 2017-02-12:
how do you currently run cli commands ?

say i wanted to install another env
`platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage`

gepd, 2017-06-18:
Done.

closed 2017-06-18 by gepd