gepd/Deviot #52: Deviot fails to create PlatformioCLI class.

ethan476, 2016-08-22:
When I attempt to open the serial monitor it fails to open with the following error:

```python
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 553, in run
    return self.run()
  File "/home/user/.config/sublime-text-3/Packages/Deviot/DeviotStarter.py", line 557, in run
    PlatformioCLI(feedback=False, callback=self.on_done).openInThread(
TypeError: __init__() got an unexpected keyword argument 'callback'
```

gepd, 2016-08-22:
Can you tell me your O.S, version of Deviot you have installed (you can see it in the status bar) and board you are working on

I've tried to reproduce the issue but I didn't have the same result

ethan476, 2016-08-22:
I believe that this is because I have not selected a serial port and Deviot is trying and failing to request that I select one.

I'm using Deviot version v1.2.2 on Ubuntu 15.10. 

I looked at the declaration for PlatformioCLI and found that there was no parameter named callback:

```python
def __init__(self, feedback=True, console=True):
        self.window = sublime.active_window()
        self.view = self.window.active_view()
        ...
       self.callback = None
        ...
```

Also I attempted to change PlatformioCLI to:
```python
def __init__(self, feedback=True, console=True, callback=None):
        self.window = sublime.active_window()
        self.view = self.window.active_view()
        ...
       self.callback = callback
        ...
```

This simple uncovers another error:

```python
Traceback (most recent call last):
  File "/opt/sublime_text/sublime_plugin.py", line 553, in run_
    return self.run()
  File "/home/user/.config/sublime-text-3/Packages/Deviot/DeviotStarter.py", line 558, in run
    'ports', process=False)
TypeError: openInThread() got an unexpected keyword argument 'process'
```

gepd, 2016-08-22:
Please let the `SerialMonitorRunCommand` class like this:

```python
class SerialMonitorRunCommand(sublime_plugin.WindowCommand):
    def run(self):
        if(not Preferences().get('id_port', False)):
            PlatformioCLI(feedback=False).openInThread('ports')
            return
        self.on_done()

    def on_done(self):
        Tools.toggleSerialMonitor(self.window)

    def is_checked(self):
        state = False
        monitor_module = Serial
        serial_port = Preferences().get('id_port', '')
        if serial_port in monitor_module.serials_in_use:
            serial_monitor = monitor_module.serial_monitor_dict.get(
                serial_port)
            if serial_monitor and serial_monitor.isRunning():
                state = True
        return state
```
in PlatformioCLI.py
```python
def __init__(self, feedback=True, console=True):
```

After that, you will prompt for a serial port, you need to make sure you have a `ino` or `cpp` file open, (I'll fix this in the next release.)
 If you keep seeing an error, select the port manually from `Deviot Menu > Serial Port(s)`

gepd, 2016-08-25:
Fixed in the last release! :smile:

closed 2016-08-25 by gepd