gepd/Deviot #53: unable to use /dev/TTYUSB0 on linux

vlad2005, 2016-08-24:
Hi!
I tried Deviot but i have an problem. When i try to build my project, after select /dev/TTYUSB0, a got this warning:
`OTA Upload are only available for Espressif (ESP8266) platforms currently`
I don't use any wireless programmer.

This is my build conf file:
`[env:pro16MHzatmega328]`
`platform = atmelavr`
`framework = arduino`
`board = pro16MHzatmega328`

`[platformio]`
`src_dir=/home/an/Arduino/temperature/final`

On usb i have connected an USB to serial adapter witch work fine with ArduinoIDE.
`#lsusb`
`Bus 004 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC`

Thanks!

gepd, 2016-08-24:
@vlad2005 I'm preparing a new release where that bug is fixed, in the meantime

Open the file `Packages/Deviot/libs/PlatformioCLI.py`  (`ST Menu > Preferences > Browse Packages`)

Replace the line 664:
```python
if(port and "COM" not in port and "esp" not in mcu):
```

for
````python
if(port and "esp" not in MCU and "tty" not in port and "COM" not in port):
```

ubidefeo, 2016-08-24:
ha!
same thing happens on Mac.
thanks for the heads up, @gepd :)

ubidefeo, 2016-08-24:
Your fix doesn't seem to work on Mac. Bummer :D

ubidefeo, 2016-08-24:
I've been playing with it a little and it seems to also always return `ATTINY85` from self.getMCU()
`selected = env_data[0]['mcu']`, I suppose

how would I revert to the previous build of DevIoT?
it worked great :)

vlad2005, 2016-08-25:
@gepd 
I tried your suggestion but got error (relevant part):
`NameError: global name 'MCU' is not defined`
I changed to this:
`if(port and "esp" not in port and "tty" not in port and "COM" not in port):`
Now i got this error:
`avrdude: ser_open(): can't open device "/dev/ttyS0": Permission denied`
My usb to serial adapter is on /dev/ttyUSBx and have permision fixed using udev.
After that started editor as superuser. Then got this error:
`08:22:36 Uploading firmware | avrdude: ser_open(): can't set attributes for device "/dev/ttyS0": Inappropriate ioctl for device`
So what to do next?

gepd, 2016-08-25:
Sorry about that guys, please [download this](https://github.com/gepd/Deviot/archive/release/V1.2.3.zip) pre-release, you shouldn't keep seeing that error.
If you find any other problem, I'll thank your report, I'm doing the last tests before the launch.

ubidefeo, 2016-08-25:
Thanks, @gepd , I'll give it a shot.
meanwhile I found that the culprit in my case is that json.loads() doesn't seem to return a dict but just leave the JSON structure intact, hence returning the first item `env_data[0]` which is 
```
{
        "id": "trinket3",
        "url": "http://www.adafruit.com/products/1500",
        "platform": "atmelavr",
        "vendor": "Adafruit",
        "rom": 8192,
        "fcpu": 8000000,
        "ram": 512,
        "mcu": "ATTINY85",
        "name": "Adafruit Trinket 3V/8MHz",
        "frameworks": [
            "arduino"
        ]
}
```

I'll let you know how this release handles it :)

ubidefeo, 2016-08-25:
ok!
had to do some cleanup but at some point it worked.

what I did:
* disable **keep temp files**
* **clean**
* delete **platformio.ini**
* quit Sublime Text 3
* launch Sublime Text 3
* select board
* select serial port
* **build**
* **upload**

it works now!
thank you for your support, @gepd  :)

vlad2005, 2016-08-25:
Thanks for your work!
1. is better to delete and Deviot file form this location:
`~/.config/sublime-text-3/Packages/User/`
2. I fixed the permision problem by adding user to groups uucp and  dialout
`sudo usermod -a -G uucp,dialout username`
3. Now just works :)

P.S. I noticed a little problem. After upgrading menu disappeared, offering only option to change the language. After I selected the "English" menu reappeared.

Big thanks!

gepd, 2016-08-25:
Thanks for your reports guys, the last platformio update changed some json structure so it broke some things.
I'm doing now the test to avoid all that reset stuff.

gepd, 2016-08-25:
The last release is available now (1.2.3) 
Any problem with the menu should be recovered automatically (If isn't recovered restart ST will do it)

Thanks for your feedback!

closed 2016-08-25 by gepd

ubidefeo, 2016-08-26:
sorry to post on this one again, @gepd 
I checked the code for the latest release, and there's one more issue about the `if` condition in `mDNScheck()`

on Mac, there's by design an alias for the `tty` port which is `cu.SERIAL_PORT_ID`, hence in the list you'll always have that port as well, but it's not handled by your `if` statement, hence causing that same OTA error.

the condition should be
```
if("esp" not in MCU and "tty" not in PORT and "cu" not in PORT and "COM" not in PORT):
```

the `cu.*` alias is the first one displayed in the list (see attachment), hence you can accidentally pick that if you don't know about the condition.

[image]

gepd, 2016-08-26:
Thanks again @ubidefeo,

I've fixed this to the next release, if you have any problem or to make sure you don't get that error replace the if by:

```python
from re import search
if(search(r"[\d{3}\.]{3}", PORT) is not None and "esp" not in MCU.lower()):
 ```

Now is checking if the port is an IP instead to check the serial ports

ubidefeo, 2016-08-27:
interesting solution :)
I think you may close the issue again
thank you