gepd/Deviot #54: Menus just disappeared during upgrade ?

tablatronix, 2016-08-24:
Did you just push an update version ?
I noticed my menus all disappeared, and I had to restart ST.

Just an FYI for UX.

gepd, 2016-08-24:
Did you only retart ST? I've released version 1.2.2 a week ago, I'll make some test to avoid that problem in the next update

tablatronix, 2016-08-24:
I think I started it then noticed it was messed up, kept using it , restarted was still messed up then after a few minutes saw my menus back again. Seems like it always takes a little while to initialize sometimes also if that is related or not sure.

gepd, 2016-08-25:
I've add a fix for that issue, now it should recover the menu automatically, if it doesn't restart ST will do

closed 2016-08-25 by gepd