gepd/Deviot #55: Upload problem

Toladar, 2016-08-26:
Hello,

I have a problem with the UPLOAD option when i upload the arduino script for the arduino Due  into the chip.

The console write :
[ Deviot 1.2.3 ] Test.ino
OTA Upload are only available for Espressif (ESP8266) platforms currently

My configuration is :
- Deviot V1.2.3
- Pio V2.11.2

Have you any solution ?

Thanks a lot !

gepd, 2016-08-27:
Hi @Toladar 
I've solved this issue for next release, but you can add the fix by yourself

Go to `ST Menu > Preferences > Browse Packages`
Open `PlatformioCLI.py` in `Deviot\libs`

Replace Line 550:
```python
if("esp" not in MCU and "tty" not in PORT and "COM" not in PORT):
```
By
````python
from re import search
if(search(r"[\d{3}\.]{3}", PORT) is not None and "esp" not in MCU.lower()):
```

garvamel, 2016-09-06:
I'm getting the same error even with the developer branch when trying to upload to a Teensy 3.2.

System:
OSX
Deviot 1.2.4
PlatformIO 2.11.2

Thanks!

gepd, 2016-09-06:
Can you tell me the name of the usb port, I've just tried that board in windows and linux and I didn't have the problem, I don't have OSX here to test, but the name of the usb can help me

garvamel, 2016-09-07:
This is the format on OSX: /dev/tty.usbmodem12345 (absolute path)
Not my specific device though I can get it tomorrow.

gepd, 2016-09-07:
Please, check the line 558 of the file `PlatformioCLI.py` in `Packages/libs` is:

````python
if(search(r"^\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}$", PORT) is not None and "esp" not in MCU.lower()):
```

I've tested the code with the path you gave me and it seems to work fine. I've check the develop branch and it should be resolved with the last fix: 
https://github.com/gepd/Deviot/blob/develop/libs/PlatformioCLI.py#L558

Just to be clear you are receiving this error: `OTA Upload are only available for Espressif (ESP8266) platforms currently` ?

garvamel, 2016-09-07:
Yes, that exact error. Line 558 is the same as you posted.
I now have the teensy at hand though, and the name of the port is longer, could that be it? I know nothing of regex, otherwise I would try to patch it myself :)

/dev/tty.usbmodem1242711

gepd, 2016-09-07:
The regex result is a normal behavior, it only match when the port is an IP.

Can you please paste this code before the line 558:
```python
        print("============")
        print("PORT:", PORT)
        print("MCU:", MCU)
        print("ENVIRONMENT:", environment)
```
After paste, restart ST and try to upload the firmware, when deviot shows the error, go to the console `Menu ST > View > Show Console` look for the previous lines and paste it here.

garvamel, 2016-09-07:
After the many restarts today, it started working. Odd to say the least.
In any case, the output was as expected I guess:
```
PORT: /dev/tty.usbmodem1242711
MCU: mk20dx256
ENVIRONMENT: teensy31
```
Sorry for the trouble! And thanks for your excellent support!

gepd, 2016-09-07:
Thanks for the info, it helps me to debug any problem. I hope to improve this in the next release!

closed 2016-09-07 by gepd