gepd/Deviot #56: Need more syntax highlighting of other data types

philectron, 2016-08-28:
Hi. I've started using Deviot, and I see that it's way better than Stino. The syntax highlighting is perfect. However, I think we need to add more data types into it.

[image]

I would love to have `boolean`, `byte`, and `word` to be **highlighted** and _italicized_ like other data types shown in the photo.

I did post a similar issue to Stino, but you said Stino is out for Deviot now. So, I want to suggest the enhancement above in the right area.

gepd, 2016-08-29:
Hi @philectron, I've added this new data types in the next release, if you want to test it, please [download](https://github.com/gepd/Deviot/archive/develop.zip) the develop branch and tell me if it's working as you mentioned above.

Just download the zip and replace the files in the Deviot Package.

philectron, 2016-08-29:
Thanks for adding more data types :100: 

I also tried using Deviot with C programming. It's wayyy better than the default C syntax highlighting. Thanks again! :+1:

closed 2016-08-29 by philectron