gepd/Deviot #58: new board defined but don't show in meniu

vlad2005, 2016-09-02:
Hi!
I try to define a new board to use with optiboot bootloader.
Edited boards.txt in "/home/dev/.platformio/packages/framework-arduinoavr/" and added this line:
`################################################################
atmega328o.name=Optiboot Arduino Pro Mini 5V 16Mhz w/ ATmega328
atmega328o.upload.protocol=arduino
atmega328o.upload.maximum_size=32256
atmega328o.upload.speed=115200
atmega328o.bootloader.low_fuses=0xff
atmega328o.bootloader.high_fuses=0xde
atmega328o.bootloader.extended_fuses=0x05
atmega328o.bootloader.file=optiboot/optiboot_atmega328.hex
atmega328o.bootloader.unlock_bits=0x3F
atmega328o.bootloader.lock_bits=0x0F
atmega328o.build.mcu=atmega328p
atmega328o.build.f_cpu=16000000L
atmega328o.build.core=arduino:arduino
atmega328o.build.variant=arduino:standard
atmega328o.bootloader.tool=avrdude
atmega328o.upload.tool=avrdude
atmega328o.build.board=AVR_PRO
`
Close subime text and reload but i can't find board in list. What is wrong?

gepd, 2016-09-02:
Isn't possible to burn a bootloader with PlatformIO 2 (The core of Deviot), but I've just found a solution with the next version ( 3 ) that will be release soon.

Please read this: https://community.platformio.org/t/burn-bootloader-to-attiny85-via-arduinoasisp/662/9

* To install PlatformIO 3 develop version go to `Deviot Menu > Options > Use Development Pio Version` 

You can confirm it's installed reading the info in the status bar. You'll see `Pio v3.0.0...` 
(The status bar is only shown when you are in a `.pde` `.ino` `.cpp` `.c` `.S` file)

* Select `Arduino Pro Mini ATmega328 (5V, 16Mhz)` from the board list and `build` the code

* Open the `platformio.ini` file from `Deviot Menu > Open Build Config File` and add `extra_script = extra_script.py` in your env, in this case `pro16MHzatmega328`

You file should looks like:

```ini
[env:pro16MHzatmega328]
platform = atmelavr
framework = arduino
board = pro16MHzatmega328
extra_script = extra_script.py
```

* Add the extra_script.py file

```python
Import("env")

env.Replace(
    SETFUSECMD='$UPLOADER $UPLOADERFLAGS -P $UPLOAD_PORT -b $UPLOAD_SPEED -Ulock:w:0xFF:m -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m',
    UPLOADBOOTCMD='$UPLOADER $UPLOADERFLAGS -P $UPLOAD_PORT -b $UPLOAD_SPEED -Uflash:w:$SOURCES:i -Ulock:w:0xFF:m'
)

bootloader_path = "/Users/antonkluev/Library/Arduino15/packages/ATTinyCore/hardware/avr/1.1.1/bootloaders/empty/empty_all.hex"
uploadboot = env.Alias(
    "uploadboot", bootloader_path,
    [env.VerboseAction(env.AutodetectUploadPort, "Looking for upload port..."),
     env.VerboseAction("$SETFUSECMD", "Setting fuses"),
     env.VerboseAction("$UPLOADBOOTCMD", "Uploading bootloader $SOURCE")])
AlwaysBuild(uploadboot)
```
You'll need to change the `bootloader_path` for the path of your hex file

The `extra_script.py` must be in the same folder as `platformio.ini`, this last file is in a temp folder. Go to `Deviot > Options > Open Build Folder` you should see a folder with the same name of your sketch, paste the python file inside that folder.

---
This last step may be a little confusing, as Deviot is not supporting this feature natively, you need to run PlatformIO CLI to end the upload

Open your terminal and navigate to the folder where the `platformio.ini` file is located.

Run the command `pio run -t uploadboot`

If all run fine, I'll start to implement this feature to the next release so it will be simpler to do it.

NOTE: If after run the command you see an error like 'not recognized', install platformio from the console with this command: `pip install -U https://github.com/platformio/platformio/archive/develop.zip`

vlad2005, 2016-09-02:
Maybe I do not understand well or I did it well. I have already installed optiboot using Arduino IDE. Now I want to install just a sketch Sublime + Deviot but into an Pro Mini with optiboot already installed. Can I use an existing board? For example Arduino Pro or Pro Mini (5V, 16 MHz) w / ATMEGA328?

gepd, 2016-09-02:
In that case you don't need to define a new board, you should be able to upload the sketch just selecting `Arduino Pro Mini ATmega328 (5V, 16Mhz)` from the board list and the serial port. :smile: 

Let me know if find any problem doing this

vlad2005, 2016-09-02:
Not working! I have this error:
`
avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/home/dev/.platformio/packages/tool-avrdude/avrdude.conf"
User configuration file is "/home/dev/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : /dev/ttyUSB0
Using Programmer              : arduino
Overriding Baud Rate          : 57600
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
`
Without change anything i can upload same sketch using arduino ide but with board defined in first post. If i try to use standard board for Pro Mini (5V 16Mhz) i got same error.
So I think it necessary to define a new board. But is not show in meniu.

gepd, 2016-09-02:
I can see the upload speed is set to `57600` as default, but in your first post it's `115200`

Try changing the upload speed:
Open the platformio ini file and add the `upload_speed` option (`Deviot Menu > Open Build Config File`). 

Your file should looks like:
```ini
[env:pro16MHzatmega328]
platform = atmelavr
framework = arduino
board = pro16MHzatmega328
upload_speed = 115200
```

Try to upload with that config and tell me if it works.

vlad2005, 2016-09-03:
I have same error.

gepd, 2016-09-03:
Please [download](https://github.com/gepd/Deviot/archive/develop.zip) the develop branch, I've add a fix to avoid this issue.

You only need to extract the zip and replace the `Deviot` folder in `Packages/Deviot` (`ST Menu >Preferences > Browse Packages`)

Let me know if that fix both issues

vlad2005, 2016-09-03:
1. Problem with upload sketch to board with optiboot using an existing board definition. It's working with suggestion to change upload speed. Now this option don't disappear from config file after each build or upload  !!!
2. Still can't add an custom board to boards.txt as indicate in first post. I don't know that is very important to define a new board, but probably is good to have.
Thanks for your hard work!

vlad2005, 2016-09-03:
After read more i found this link on platformio website:
[http://docs.platformio.org/en/latest/platforms/creating_board.html](url)
Here explain how to create new board. I try'it to create an json file named optiboot.json inside boards folder in my home directory as is indicated in link above.
Here is my json file:
'\"optibootPro16MHzatmega328\":{\"vendor\": \"Arduino\", \"name\": \"Arduino Pro Mini Optiboot ATmega328 (5V, 16 MHz)\", \"platform\": \"atmelavr\", \"url\": \"http://arduino.cc/en/Main/ArduinoBoardProMini\", \"upload\": {\"maximum_ram_size\": 2048, \"maximum_size\": 32256, \"protocol\": \"arduino\", \"speed\": 115200, \"require_upload_port\": true}, \"frameworks\": [\"arduino\"], \"build\": {\"core\": \"arduino\", \"mcu\": \"atmega328p\", \"f_cpu\": \"16000000L\", \"extra_flags\": \"-DARDUINO_ARCH_AVR -DARDUINO_AVR_PRO\", \"variant\": \"eightanaloginputs\"}}'
But i don't have any success.
P.S. It's possible to use command like this:  platformio boards to view existing boards?
Anyway, from menu, optiboot board defined by me don't appear.

gepd, 2016-09-04:
Your file must be well formated:

```json
{
    "optibootPro16MHzatmega328": {
        "build": {
            "core": "arduino",
            "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_PRO",
            "f_cpu": "16000000L",
            "mcu": "atmega328p",
            "variant": "eightanaloginputs"
        },
        "frameworks": ["arduino"],
        "name": "Arduino Pro Mini Optiboot ATmega328 (5V, 16 MHz)",
        "platform": "atmelavr",
        "upload": {
            "disable_flushing": true,
            "maximum_ram_size": 2048,
            "maximum_size": 32256,
            "protocol": "arduino",
            "require_upload_port" : true,
            "speed": 115200
        },
        "url": "http://arduino.cc/en/Main/ArduinoBoardProMini",
        "vendor": "Arduino"
    }
}
```
1. Save the file, as explained in the PlatformIO documentation
2. Remove the file located in: `Packages\User\Deviot\Preset\platformio_boards.json` and restart ST 

When ST start again will generate the file automatically again, now you will see the new board in the list

vlad2005, 2016-09-04:
For some strange reason i don't see new board in list. But if i change build config file to this:
```javascript
[env:pro16MHzatmega328]
platform = atmelavr
framework = arduino
board = optibootPro16MHzatmega328
[platformio]
src_dir = /home/andra/Arduino/porti/Gate
```
then was able to build and upload sketch to my board. So new board is know, but not show in menu.

gepd, 2016-09-04:
Good to hear, I forgot to mention ST loads all the plugins first, so it take some seconds to re-generate the file again.

I'll considerate to show a text in the console when a new board is found

vlad2005, 2016-09-04:
Will be nice!
Thanks for help!

closed 2016-09-09 by gepd

gepd, 2016-09-18:
INFO: Since PlatformIO 3 your JSON file must be formated in this way:

```json
{
    "build": {
        "core": "arduino",
        "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_PRO",
        "f_cpu": "16000000L",
        "mcu": "atmega328p",
        "variant": "eightanaloginputs"
    },
    "frameworks": ["arduino"],
    "name": "Arduino Pro Mini Optiboot ATmega328 (5V, 16 MHz)",
    "platform": "atmelavr",
    "upload": {
        "disable_flushing": true,
        "maximum_ram_size": 2048,
        "maximum_size": 32256,
        "protocol": "arduino",
        "require_upload_port" : true,
        "speed": 115200
    },
    "url": "http://arduino.cc/en/Main/ArduinoBoardProMini",
    "vendor": "Arduino"
}
```
The name of the environment will be get from the file name, in this case, it should be: `optibootPro16MHzatmega328.json`