gepd/Deviot #59: persisting/new issues with mDNSCheck()

ubidefeo, 2016-09-03:
hi @gepd 
I'm still having issues on Mac with how detection of port and board is handled in `mDNSCheck()`.

I've placed some debugging code in there to see where it halts, and it's clearly the regExp failing something.
Unluckily I'm clueless about regExp, so I don't know what fails there.
sure thing is if I execute `search(r"[\d{3}\.]{3}", PORT)` with `PORT='/dev/tty.usbserial-A900adQQ'` (my USB Serial port) in a Python console I get nothing.

I solved by just commenting out the `return false` so I can at least move on and work, but I think a better solution needs to be found here.
The previous fix based on `if` worked, but maybe not for everyone.
In my opinion, searching for the presence of `tty`, `cu.`, `COM` (case insensitive) followed by a number might fix it for all the cases I'm aware of.

thanks for your attention
u.

gepd, 2016-09-03:
Sorry about that @ubidefeo 

Can you try replacing the if for this line?

```python
if(search(r"^\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}$", PORT) is not None and "esp" not in MCU.lower()):
```
(Remember to restart ST after the change)

It should not show the error this time

ubidefeo, 2016-09-03:
worked like a charm, mister :)
I'm gonna have to learn some regExp at some time

thank you

ubidefeo, 2016-09-03:
by the way, about having to restart ST all the time, isn't there some way to tell ST to just reload a plugin?
it's silly that if you're patching/testing you have to restart the whole shebang.
I haven't looked into plugin dev yet because my Python skills are quite poor :)

gepd, 2016-09-03:
To avoid restart everytime I made a change I use this plugin: https://github.com/randy3k/PackageReloader
It's very useful!

ubidefeo, 2016-09-03:
ace!

closed 2016-09-03 by gepd