gepd/Deviot #60: How upload code without write eeprom

vlad2005, 2016-09-09:
Where i can change some options so when upload new code eeprom not be erased. I know that i need to remove parameters for fuses but don't know exactly where need to do that.

Thanks!

gepd, 2016-09-09:
Look this info: http://docs.platformio.org/en/stable/projectconf.html#build-flags

You should add the flag(s) in the platformio.ini just as you did with the board definition. Try that and let me know if you have any problem

vlad2005, 2016-09-09:
I don't think that is something related to build flags. Don't see any reference to fuses in build flags documentation from platformio.
I think that solution is to use "extra_script" option.

gepd, 2016-09-09:
Yes the  solution should be extra_script, look this answer https://community.platformio.org/t/burn-bootloader-to-attiny85-via-arduinoasisp/662/9 

You should add only this lines, and modify it as you want
```python
Import("env")

env.Replace(
    SETFUSECMD='$UPLOADER $UPLOADERFLAGS -P $UPLOAD_PORT -b $UPLOAD_SPEED -Ulock:w:0xFF:m -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m',
    UPLOADBOOTCMD='$UPLOADER $UPLOADERFLAGS -P $UPLOAD_PORT -b $UPLOAD_SPEED -Uflash:w:$SOURCES:i -Ulock:w:0xFF:m'
)
```

Add the extra_script option in the platformio.ini with the path of your python file

vlad2005, 2016-09-10:
Ok, thanks!
Meanwhile I checked more and it seems that eeprom is not write when upload new code.
But above solution it is good to know!

closed 2016-09-15 by gepd