gepd/Deviot #61: Assuming OTA upload

nielsclausen, 2016-09-09:
After upgrading, I cannot upload anything? As I see it, it assumed I try to use OTA uploading, but I have never used that or even seen that before.

The message is simply: 
```
[ Deviot 1.2.4 ] DevIOTtest.ino
OTA Upload are only available for Espressif (ESP8266) platforms currently
```

[image]

When I just build the project, all is well:
```
[ Deviot 1.2.4 ] DevIOTtest.ino
20:45:36 Building the project | Processing...
SUCCESS | it took 1.10s
```

The blink example, used for this report:
```
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
} 
```

gepd, 2016-09-09:
Sorry about that, I've just release version 1.2.5 with a fix for that issue, it will take some hours until package control get the last release, you can check that here: https://packagecontrol.io/packages/Deviot

If you see `VERSION 1.2.5` is because it's done, restart ST should update Deviot if don't you can do:
`Preferences > Package Control > Type: Package Control: Upgrade Package` and select Deviot from the list

nielsclausen, 2016-09-09:
Alright! I'll just wait. Keep up the good work!!!

gepd, 2016-09-09:
Thanks @nielsclausen 
If you don't want to wait, just download [this file](https://github.com/gepd/Deviot/archive/master.zip) and replace the files in the package folder (`ST Menu > Preferences > Browse Packages`)

gepd, 2016-09-09:
If you keep with the problem please re-open this issue.

closed 2016-09-09 by gepd

nielsclausen, 2016-09-09:
Alright, I'll try again Monday.