gepd/Deviot #62: Console Error after build. Doesn't affect anything though.

garvamel, 2016-09-13:
Hello!

I'm in OSX, with ST3, and Deviot developer branch.

After build, ST console shows the following error:

```
Exception in thread Thread-265:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/gianpaolo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/PlatformioCLI.py", line 280, in build
    Tools.ERRORS_LIST.extend(Tools.highlightError(exec_view))
  File "/Users/gianpaolo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Tools.py", line 750, in highlightError
    if(string_before in current_line):
UnboundLocalError: local variable 'string_before' referenced before assignment
```

The only thing I can notice, is that if building fails or has warnings for some reason, the issues displayed lack capitalized letters, even if code has them.

Hope it helps!

garvamel, 2016-09-13:
Update:

Updated PlatformIO and restarted sublime. Caps are back but error persists.

gepd, 2016-09-13:
Thanks for you report @garvamel I had noted that error some days ago. it's caused because the highlight function. You shouldn't have problem to build or upload your firmware, I'm working to fix and improve that feature in the next release.

garvamel, 2016-09-14:
The second issue I noticed is that It only randomly enables the function  "click warning/error and go to line". Most of the time it doesn't work.

Thanks!

gepd, 2016-09-14:
Yes, it's related to the same issue, I'll let you know when all this be fixed

gepd, 2016-09-16:
Can you try the [develop](https://github.com/gepd/Deviot/tree/develop) branch and check if the go to line error is working now?

garvamel, 2016-09-16:
Seems to work perfectly!

gepd, 2016-09-16:
Thanks for your report!

closed 2016-09-16 by gepd