gepd/Deviot #63: build_flags error in platformio.ini

garvamel, 2016-09-14:
I have the following build flags set in my .ini file:

`build_flags = -larm_cortexM4l_math -lm -Wl,-u,_printf_float`

But every time I upload, the commas get expanded and a space is added leaving it like this:

`build_flags = -larm_cortexM4l_math -lm -Wl,    -u,    _printf_float` (extra spaces added for readability)

Which is wrong syntax (I have used these flags before, with Deviot 1.2.4 and PlatformIO 2.11.2) and therefore build fails, with the following error:

```*** [.pioenvs/teensy31/firmware.elf] Implicit dependency `/Users/gianpaolo/.platformio/platforms/teensy/builder/_printf_float' not found, needed by target `.pioenvs/teensy31/firmware.elf'.```

System is OSX 10.11.6
Deviot 1.2.5
PlatformIO 3.0.1

Thanks!

gepd, 2016-09-14:
I've push a fix in the developer branch, can you check if it solves your problem?

garvamel, 2016-09-16:
It still gets expanded :(

gepd, 2016-09-16:
Can you check if [this](https://github.com/gepd/Deviot/commit/bef5d6c7191b5bb41ba3f0d6ce7ec41eaac962e6) commit is in your code?

garvamel, 2016-09-16:
It wasn't. After adding it, error is fixed. Building process hangs though, but without any errors (with verbose build).

gepd, 2016-09-16:
Let's check if it's Deviot or PlatformIO problem.

- Go to `Deviot Menu > Options > Open Build Folder` and look the folder with the same name of your sketch

- Open the terminal and go to the folder found in the previous step

- Run this command `"script path\pio" run -t upload`

The `script path` should be the full path of the folder located in `Packages\User\Deviot\penv\bin`

After run that command tell me if the behavior is the same

garvamel, 2016-09-16:
`Open Build Folder: Deviot can't be found`

Even if I change the build folder and try to build again, or open the build folder after selecting, same issues arise. I'll install developer branch again and try.

garvamel, 2016-09-16:
Same issues after manual re install of developer branch.

gepd, 2016-09-16:
I'll check that option. It opens the folder `/tmp/Deviot`

If you can run the command in the console I'll thanks if you copy the output here

garvamel, 2016-09-16:
That one doesn't exist either, since build hangs.

gepd, 2016-09-16:
I've create a test project

platformio.ini
```ini
[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
build_flags = -larm_cortexM4l_math -lm -Wl,-u,_printf_float
```
And a simple sketch
```c++
void setup()
{
    
}
void loop()
{
    
}
```
I can build that sketch normally, If you can too, please upload the sketch are you working on

garvamel, 2016-09-21:
I'm sorry I wasn't able to reply earlier. I found the issue though.

Sketch builds fine without verbose build. When ticking the option "Full Verbose Output When Build", the build process hangs with the same ST console error I pasted beforehand:

```
Exception in thread Thread-11:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/gianpaolo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/PlatformioCLI.py", line 276, in build
    CMD.runCommand(command, "built_project_{0}")
  File "/Users/gianpaolo/Library/Application Support/Sublime Text 3/Packages/Deviot/libs/Commands.py", line 113, in runCommand
    output = process.stdout.readline()
  File "./python3.3/encodings/ascii.py", line 26, in decode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xcc in position 48: ordinal not in range(128)
```

gepd, 2016-09-22:
I suspect about how fix the problem, I tried with some codes but none of them give that error.

Can you upload the code and the `platformio.ini` file used?

garvamel, 2016-09-22:
Fixed it. My path had a character outside the charset it is testing for, specifically an `ó` (my native language is Spanish), therefore the path couldn't be resolved.

It was something like this:

/garvamel/Dev/Flujómetro/...  -> error

changed it to:
/garvamel/Dev/Flujometro/... -> no error and successful build process.

gepd, 2016-09-23:
Thanks for the report. I've reproduced the issue, I'll try to find a solution without manually changing the name

garvamel, 2016-09-26:
Awesome, thanks!

gepd, 2017-09-08:
This issue was finally fixed with: https://github.com/gepd/Deviot/commit/ef012b106773689536287d7b3db338ccdf42d5ae

There not need to change anyting

Re-open if there is any issue related to this

closed 2017-09-08 by gepd