gepd/Deviot #64: python2 and platformio not recognized (Arch Linux)

Jakeler, 2016-09-15:
"Deviot needs python2.7" popups all the time, although i have `python2` + `python2.7` installed and in the PATH... but Deviot searches only for `python`, which is linked to `python3.5` on Arch. It works when i change this symlink to 2.7  (no permanent solution, breaks a lot other stuff).

After that Deviot says "PlatformIO is not installed" and downloads everything. Even though i also had `platformio` and `pio` already installed in /usr/bin.

gepd, 2016-09-15:
Hi @Jakeler, 
to understand, normally to run pyton 2.7 you use `python2 command` is that right? I'll check how to easily change that, it shouldn't be a big problem

Deviot checks if PlatformIO is installed in the machine before to procced with the new install, I'll check again the default paths to see if that is the problem

Jakeler, 2016-09-15:
Yes, that is right and `#!/usr/bin/env python2`  in scripts.

gepd, 2016-09-15:
Let's make some test

- First [download](https://github.com/gepd/Deviot/archive/develop.zip) the `develop` branch and replace the files in `Packages/Deviot`

- Locate the file in `Deviot/libs/Install.py` and change the line 49 to:
```python
self.python = os.path.join(self.env_bin_dir, 'python2')

```

 - Locate the file in `Deviot/libs/Commands.py` and change the line 33 to:
```python
self.python = os.path.join(env_bin_dir, 'python2')

```

After this, Deviot should always use `python2` to run the commands.

**To the second issue:**

 - Locate the file in `Deviot/libs/Tools.py` and change the line 210 to:
```python
default_path = ["/usr/bin", "/usr/local/bin", "/usr/bin/env"]

```

Remove the folder `Deviot` located in `Packages/User` and restart ST. After this, the plugin should works without any extra change, and when the setup start it should say `PlatformIO is installed in your system, you can enjoy Deviot now!`

Let me know if that fixes works as I explained. If it does I'll add some new options to the next release.

Jakeler, 2016-09-15:
That did not fix it unfortunately, i still get the same popup... Also the 2. issue (with the changed symlink).

btw, wouldn't it be easier if you put these experimental changes just into a other branch? (so you don't have to explain what i have to edit)

gepd, 2016-09-16:
Let's use this branch to do our test: https://github.com/gepd/Deviot/tree/newpython

Download it and test again

Jakeler, 2016-09-16:
Okay, i fiddled around a while with this and i am still a bit confused what Sublime/Package Control does. It seems that you can't just change files (changes nothing, even with a restart), then there was a `Backup` folder with many `Deviot` folders in it that affected the result...
Anyway, i verified it in a fresh VM, it works now with python2! 

But the issue with platformio is still present.

gepd, 2016-09-16:
I don't know why did you have problem, you can replace the files manually, I've do it without problem in some of my VMs

But that is a progress. I'll check again why isn't recognizing platformio previously installed.

gepd, 2016-09-17:
Please, Install `newpython` branch again, I've a new commit to fix the issue with PlatfomIO previously installed

Jakeler, 2016-09-17:
`PlatformIO is installed in your system, you can enjoy Deviot now!`
Thanks a lot, works perfectly!

gepd, 2016-09-17:
Excellent, I still have to tune some things. I'll let you know when it's complete, but meanwhile you can use that version

jwhendy, 2016-11-02:
Is this implemented in `master` or whatever default Package Control installs? I had some issues with `stino` after an Arch upgrade and stumbled on this so I thought I'd try it out.

I got the `python` errors as well and just decided to symlink `/usr/bin/python -> /usr/bin/python2.7` for a quick fix. Worked fine, but prompted me to ask based on the above if this should have been required. If this is integrated into `master`, it didn't work for me on the first try. I don't see any references on the readme about it or how one would know they might run into this issue.

Thanks!

gepd, 2016-11-03:
I still not release the new version with this feature, but it's in the `newpython` branch. You can [Download](https://github.com/gepd/Deviot/tree/newpython) it and replace the files in `Packages/Deviot`

The reference to the symlink in the [install file](https://github.com/gepd/Deviot/blob/newpython/libs/Install.py#L49) in case you need it

gepd, 2017-06-06:
After some time away from this repository, I'm back to work in this issues, I hope to have all that features merge any time soon.

gepd, 2017-06-20:
I've improved this feature in the in the last version of the plugin, it's not released yet but you can help me to test it, more information here: https://github.com/gepd/Deviot/issues/117

closed 2017-06-27 by gepd