gepd/Deviot #65: Automatic removal of variables in platformio.ini

ubidefeo, 2016-09-16:
hi Gepd

as I get more into Platform.io, I've started to customise the .ini file to my likings,
and yesterday I realised that my `upload_port` variable was being removed on upload.
I went through the plugin code and found this:
```
ENV = INIFILE[ENVIRONMENT]
rm = ['upload_protocol', 'upload_flags', 'upload_speed', 'upload_port']

# remove previous configuration
    if(rm[0] in ENV):
        for line in rm:
            if(line in ENV):
                ENV.pop(line)
```

I think that when using native, these changes should not happen.

u.

gepd, 2016-09-16:
That's weird, this line `if(rm[0] in ENV):` check if `upload_protocol` is in the platformio.ini, if isn't doesn't remove anything. I've check with a simple sketch and it seems not to remove that option.

Can you paste here your platformio.ini

have you instaled last version of Deviot (1.2.5)?

gepd, 2016-09-18:
I've introduced the option to recognize environments manually added in platformio.ini. Now it's listed in the same list of the others environments

gepd, 2016-09-26:
@ubidefeo did you keep having this issue?

ubidefeo, 2016-09-27:
I decided to uninstall the package and manually add it as a .git repo so I can switch branches and run some tests or, you never know, submit a patch if I get my Python together :)

Anyhow...
Using the develop branch the Environments list properly, but I noticed that if the env name does not match a compatible board nothing happens during upload.
The console stays empty as well.
If I change the name of the env to one of the supported boards then it works and uploads.

Also it seems that the `upload_port` set manually in the env is ignored in favour of the manually set one :)

tablatronix, 2016-12-09:
I have also noticed this, will also test

gepd, 2017-06-30:
The last issue is fixed in Deviot 2.0

closed 2017-06-30 by gepd