gepd/Deviot #66: error compile when define ISR routine for external interrrupt

vlad2005, 2016-09-17:
I got an error when i try to implement in ISR for external interrupt:
```
09:30:12 Building the project | Processing...
.pioenvs/optibootPro16MHzatmega328/libFrameworkArduino.a(WInterrupts.o): In function `__vector_2':
collect2: error: ld returned 1 exit status
09:30:16 ERROR | it took 4.49s
```
Here is implementation
```javascript
#ifdef EXTERNAL_INTERRUPT
ISR(INT1_vect) {
  CounterControl::GATE_ISR();
};

void CounterControl::GATE_ISR() {

}
#endif
```

gepd, 2016-09-17:
Which board are you using `megaatmega2560`? 
If don't, please upload your platformio.ini

vlad2005, 2016-09-17:
platformio.ini
```config
[env:optibootPro16MHzatmega328]
platform = atmelavr
framework = arduino
board = optibootPro16MHzatmega328
[platformio]
src_dir = /home/dev/Arduino/test/Gate
```
here is board config
```json
{
    "optibootPro16MHzatmega328": {
        "build": {
            "core": "arduino",
            "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_PRO",
            "f_cpu": "16000000L",
            "mcu": "atmega328p",
            "variant": "eightanaloginputs"
        },
        "frameworks": ["arduino"],
        "name": "Arduino Pro Mini Optiboot ATmega328 (5V, 16 MHz)",
        "platform": "atmelavr",
        "upload": {
            "disable_flushing": true,
            "maximum_ram_size": 2048,
            "maximum_size": 32256,
            "protocol": "arduino",
            "require_upload_port" : true,
            "speed": 115200
        },
        "url": "http://arduino.cc/en/Main/ArduinoBoardProMini",
        "vendor": "Arduino"
    }
}
```

vlad2005, 2016-09-17:
Remove package, delete all files and reinstall deviot with platformio 3.0.1. I have same problem.

vlad2005, 2016-09-17:
Solved!
It's seem that EXTERNAL_INTERRUPT is used in some libraries. I change name to EXT_INTERRUPT and not project compile without error.
It is not a issue relating to Deviot package. 
I apologize for the inconvenience!

gepd, 2016-09-17:
Good to hear!

As recommendation, I realized your JSON file hasn't the right format, in PlatformIO 3 it should be like this:

```json
{
    "build": {
        "core": "arduino",
        "extra_flags": "-DARDUINO_ARCH_AVR -DARDUINO_AVR_PRO",
        "f_cpu": "16000000L",
        "mcu": "atmega328p",
        "variant": "eightanaloginputs"
    },
    "frameworks": ["arduino"],
    "name": "Arduino Pro Mini Optiboot ATmega328 (5V, 16 MHz)",
    "platform": "atmelavr",
    "upload": {
        "disable_flushing": true,
        "maximum_ram_size": 2048,
        "maximum_size": 32256,
        "protocol": "arduino",
        "require_upload_port" : true,
        "speed": 115200
    },
    "url": "http://arduino.cc/en/Main/ArduinoBoardProMini",
    "vendor": "Arduino"
}
```

Rename the json file as: `optibootPro16MHzatmega328.json` That way you will avoid any problem

vlad2005, 2016-09-18:
Yes, is truth! After i make update i see that not working with old json and need to do exactly what u say (after many try's). Maybe is better to make an update to [this](https://github.com/gepd/Deviot/issues/58) thereat to help and other users.

closed 2016-09-18 by gepd