gepd/Deviot #67: Building success, but upload fails (ArchLinux)

ryukinix, 2016-09-19:
Well, I have all troubles exactly how was be described here #64, but I really have need remove my previously instalation of plataformio from AUR because he just blame on pip install about don't support `python3`. Maybe inside of the `Deviot` it are calling `pip` instead `pip2`. `pip` means `python3` because python3.5 is the default.

BTW, the problem is another, after fix all that things, unfortunelly, I have a new problem. After setup my board an serial port... great, the code compiles! But during upload, just fails :(

On log of Sublime I have really insane stuffs:

```
/bin/sh: /home/lerax/.config/sublime-text-3/Packages/User/Deviot/penv/bin//home/lerax/.config/sublime-text-3/Packages/User/Deviot/penv/bin/python: No such file or directory
```

Why that path are duplicated? (I'm using your `newpython` branch, BTW).

gepd, 2016-09-19:
When Deviot install PlatformIO automatically, it calls pip but in a virtualenv so I guess I wouldn't be a problem, in the case of the `newpython` branch you only need to make sure when `python2` is called, python 2.7 is used.

What board are you using?

Go to `Deviot Menu > Options > Full Verbose Output When Build` and paste here any extra info about the error

ryukinix, 2016-09-19:
I'll do this. I tried build with `Arduino Nano AT326` and `Arduino Mega AT2560`.

ryukinix, 2016-09-19:
```
[ Deviot 1.2.5 ] RandomMusic.ino
03:53:57 Initializing the project | Processing...

The current working directory /tmp/Deviot/RandomMusic will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in /tmp/Deviot/RandomMusic
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Project has been successfully initialized!
Useful commands:
03:53:59 Uploading firmware | `platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
[Mon Sep 19 03:54:00 2016] Processing nanoatmega328 (platform: atmelavr, board: nanoatmega328, framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting RandomMusic.ino
Collected 25 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs/nanoatmega328/src/RandomMusic.ino.o
Archiving .pioenvs/nanoatmega328/libFrameworkArduinoVariant.a
Indexing .pioenvs/nanoatmega328/libFrameworkArduinoVariant.a
Compiling .pioenvs/nanoatmega328/FrameworkArduino/CDC.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/HardwareSerial.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/HardwareSerial0.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/HardwareSerial1.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/HardwareSerial2.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/HardwareSerial3.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/IPAddress.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/PluggableUSB.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/Print.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/Stream.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/Tone.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/USBCore.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/WInterrupts.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/WMath.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/WString.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/_wiring_pulse.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/abi.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/hooks.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/main.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/new.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/wiring.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/wiring_analog.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/wiring_digital.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/wiring_pulse.o
Compiling .pioenvs/nanoatmega328/FrameworkArduino/wiring_shift.o
Archiving .pioenvs/nanoatmega328/libFrameworkArduino.a
Indexing .pioenvs/nanoatmega328/libFrameworkArduino.a
Linking .pioenvs/nanoatmega328/firmware.elf
Checking program size .pioenvs/nanoatmega328/firmware.elf
text	   data	    bss	    dec	    hex	filename
1034	      0	      9	   1043	    413	.pioenvs/nanoatmega328/firmware.elf
Building .pioenvs/nanoatmega328/firmware.hex
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/nanoatmega328/firmware.hex
/home/lerax/.platformio/packages/toolchain-atmelavr/bin/avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
*** [upload] Error 127
========================== [ERROR] Took 4.02 seconds ==========================
03:54:04 ERROR | it took 4.23s
```

ryukinix, 2016-09-19:
ẀOW, would be just ncurses?

ryukinix, 2016-09-19:
Hmms, seems something with platformio. I'm checking that https://github.com/platformio/platformio/issues/291

ryukinix, 2016-09-19:
Great, I fixed with that nice debugging output (i don't see the menu before xD, sorry). The solutions would a little different whose is proposed in that last issue i commented here. The command to fix that (for other people whose can has the same problem as I had) is:

`ln -s -f /usr/bin/avrdude ~/.platformio/packages/toolchain-atmelavr/bin/avrdude`

Thanks for the support. I wanna help you in soon. This is really a awesome plugin!!!

gepd, 2016-09-19:
Excellent!

Any help will be always well received

closed 2016-09-19 by gepd