gepd/Deviot #68: check for platformio update

vlad2005, 2016-09-22:
Hi!
If press Shift+Alt+P to check for platformio upgrade, or via menu, i see an message in console
```
Checking for PlatformIO updates...
```
but then nothing. 
P.S. I see that u want to implement platformio command line. Will be nice!

gepd, 2016-09-22:
Can you paste here the output of the console (`View > Console`) after run that command?

Yes, I'm currently working in a way to have the command line. I hope to have something soon, will be great to have feedback to keep improving it.

gepd, 2016-09-22:
That is the Deviot console, from `ST Menu > View > Console` you should get a different output

vlad2005, 2016-09-23:
Today, when we launched the editor offered me the opportunity to make updates through a message in a window control. I accepted and made update to platformio 3.1.0. Now, when check for update in console appears this message:
[Deviot 1.2.5]
Checking for updates ... PlatformIO
PlatformIO is up to date (3.1.0)
But until to update, appeared only the first two lines. and nothing more. Maybe it was a problem with the current version check.

gepd, 2016-09-23:
I've found the bug, next time you won't have that problem

closed 2016-09-23 by gepd

vlad2005, 2016-09-23:
Ok, thanks!