gepd/Deviot #69: question: how to ota ?

tablatronix, 2016-09-24:
I am probably missing something.

I am not sure how to get OTA working, If i click serial port enter ip, and enter it, nothing happens.
Next time I do upload it just asks for serial port again.

If i manually add `upload_port` to .ini, same thing.

am i missing something obvious ?

tablatronix, 2016-09-25:
I do have this in my log

```
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/DeviotStarter.py", line 596, in on_done
    Menu().createSerialPortsMenu()
AttributeError: 'Menu' object has no attribute 'createSerialPortsMenu'
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/DeviotStarter.py", line 596, in on_done
    Menu().createSerialPortsMenu()
AttributeError: 'Menu' object has no attribute 'createSerialPortsMenu'
Traceback (most recent call last):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot/DeviotStarter.py", line 596, in on_done
    Menu().createSerialPortsMenu()
AttributeError: 'Menu' object has no attribute 'createSerialPortsMenu'
```

gepd, 2016-09-26:
Can you try the developer branch and tell me if the problem is present there too?
I'm not with much time this week, but I've use OTA upload without problem few days ago

tablatronix, 2016-09-26:
No difference, but after I entered the upload_port in .ini file it worked once, then it asked for serial port again. If I do manual enter ip , it does nothing.

And i only ever remember seeing the network port show up nonce in that list.
My arduino IDE shows network ports automatically and reliably, so no problem with my boards or network.

I am on mac.

gepd, 2016-10-06:
I've made a few change in the code and you should be able to use OTA now.

Now it shows the device in the serial ports list.

To make sure all works please re-install the develop branch  and remove the `Deviot` folder in `Packages/User`.

tablatronix, 2016-10-07:
I will test

tablatronix, 2016-12-10:
will have to wait on #72 to test

airdrummingfool, 2017-05-20:
I had this issue as well on MacOS 10.12.4 with Sublime Text 3 and Deviot 1.2.5. I couldn't see my ESP in the serial port list or manually add an IP address (the console would print `AttributeError: 'Menu' object has no attribute 'createSerialPortsMenu'`).

Based on your comment above I removed 1.2.5 and installed the `develop` branch version. Changing to the `develop` version fixed the error I was seeing in the console and allows me to manually set an IP address to program OTA. However, the network port still doesn't reliably show up in the serial port list. I've seen it a few times, but anytime I see it, it's gone the next time I open the serial port list (I've opened the serial port list ~30 times, and it was listed ~3 times). It always shows up fine in the Arduino IDE.

Even though the network serial port doesn't show up in the serial port list, I am now able to program OTA by using "Add IP Manually".

closed 2017-06-25 by gepd