gepd/Deviot #70: new odd behavior in 1.25+ change serial port triggers upload

tablatronix, 2016-09-26:
I noticed recently , when I change a serial port, it automatically starts uploading the current project.
This is very annoying, not sure what is causing it.

tablatronix, 2016-09-26:
It also opens serial monitor as well.
I can see where this would be a feature, I am just not sure how it could work though.

gepd, 2016-09-26:
Isn't a feature, I'm working to solve all the bugs you reported, I hope to have a solution today

gepd, 2016-09-26:
This bug should be fixed with https://github.com/gepd/Deviot/commit/d66b36425293da4cdfc4e39d07bc83acbbd4acc9 let me know if you keep getting this behavior

airdrummingfool, 2017-05-20:
I had this issue on 1.2.5, but it appears to be fixed on the `develop` branch.

closed 2017-06-30 by gepd