gepd/Deviot #71: default serial port speeds ?

tablatronix, 2016-09-26:
How can we modify the default serial monitor drop down speeds ?
I want to add 
74880 ( esp8266 bootloader )
921600, commonly used

gepd, 2016-09-26:
Open the file `menu_main.json` located in `Packages\Deviot\Preset` and adds the desired speed, after change and save the change go to `Deviot Menu > Language` and select the same language in use. Accept the warning to restart. After restart the new speed will appear.

tablatronix, 2016-09-26:
works, thanks!

tablatronix, 2018-06-22:
Has this moved, I do not see serial baud in that file anyhmore.

gepd, 2018-06-26:
Yes, now it's in: https://github.com/gepd/Deviot/blob/develop/libraries/quick_menu.py#L519

gepd, 2018-06-26:
I've added `74880` as default

tablatronix, 2018-06-26:
ahh so its hardcoded again, thanks

closed 2018-06-26 by tablatronix